prusa3d/PrusaSlicer #12055: Request to add a new printer profile Creality CR-10 SE

Vrrkhm, 2024-01-12:
Hello, is there an option for you to add a Creality CR-10 SE printer profile to PrusaSlicer? I really like to use your slicer.

SkipTheRed, 2024-02-09:
I found this.  Might be a good starting point for PrusaSlicer.
https://auroratechchannel.com/download/CR10SE.zip

Zarin-Igor, 2024-04-04:
Hello, thanks for adding CR 10 SE to the new version, but unfortunately it doesn’t work, the following error pops up. I also wanted to ask if it is possible to directly send a function for printing from the program. Thank you.

[image]

Jan-Soustruznik, 2025-08-29:
Third-party printer profiles are created by the community. If a printer profile isn’t implemented in PrusaSlicer, you can create the profile yourself or add it as a config bundle. You can create PR at https://github.com/prusa3d/PrusaSlicer-settings-non-prusa-fff (Creality bundle already exists there).

I closed this issue as duplicate of https://github.com/prusa3d/PrusaSlicer/issues/14499

closed 2025-08-29 by Jan-Soustruznik