prusa3d/PrusaSlicer #12061: Probleme exportatopn gcode suite mise a jour vers 2.7.1

ironwarthog1, 2024-01-14:
### Description of the bug

je ne peut pas aller a la fin du découpage car une erreur me dis que je ne peut pas exporter le Gcode et uniquement depuis la mise a jour vers 2.7.1, avec la version 2.6.1 aucun problème, j'étais déjà revenu a l'ancienne version il y a quelques semaine.
pensant le problème résolu j'ai voulu faire la mise a jour hier et le problème est toujours la.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

window pro

### Printer model

fl sun super racer

ironwarthog1, 2024-01-14:
et je ne trouve pas l'adresse export to dans mon disque dur

neophyl, 2024-01-14:
its Not "Input_filename", it is "input_filename_base" 
Please see https://help.prusa3d.com/article/list-of-placeholders_205643#custom-g-code-specific-placeholders

closed 2026-09-29 by prusaslicer-bot[bot]