prusa3d/PrusaSlicer #12062: Improve multi-material first interface layer

Hausmeister486, 2024-01-14:
It's about a function that was implemented better in Bambu Studio than in Prusa Slicer 2.7.1
Printer: Prusa XL5
Scenario:
Print object made of PLA (extruder 1)
PLA support structure (extruder 1)
Support Interface layer (extruder 2) with PETG

Top contact Z distance: 0 (soluble) 
Pattern: Rectilinear
Pattern space: 0mm
Topinterface layer: same as top

In the Prusa Slicer, a support structure made of PLA with a rectilinear pattern is printed in the same way up to the interface layer.
Then 3 Inteface PETG layers are printed. 
Here, the first inteface layer is printed in PETG and has poor adhesion on PLA.
There is no option to change this.

see picture 1 , first interface layer in PETG (Prusa Slicer 2.7.1)

[image]

Bambu Studio:
In Bambu Studio, under the first interface layer in PETG, the last layer of the support structure is printed like an interface layer, but in the material of the support structure (PLA) and only then are the other real interface layers printed with PETG. This gives the first PETG interface layer more strength or adhesion and the other layers are printed more reliably. This is standard in Bambu Studio and cannot be changed.

picture 2, last Support layer (created in Bambu Studio) 
Here, the last support layer is still printed in PLA and the following interface layer in PETG find a solid layer for better adhesion
[image]

picture 3
[image]

Proposed solution for the Prusa Slicer:
A good solution might be a selection option that allows you to choose whether the first interface layer should be printed with Extruder1 or Extruder2

Best regards,

argosforge, 2024-03-28:
This change is critical to making multi-material supports work - it just simply DOES NOT work (adhere) otherwise.

We need this to take full advantage of the XL multi-material handling capabilities

morphias2004, 2024-05-10:
Prime example of this issue:

[image]

[image]

Hausmeister486, 2024-06-05:
Reminder

Racemade, 2024-06-10:
UP!
Please, it would be awesome! Actually it‘s useless!!!

Hausmeister486, 2024-07-07:
It would be great if the number of interface layers could be freely selected and the extruder could be chosen separately for each interface layer.

This would also make complex support interface structures conceivable, e.g. three different materials, with the third filament as a special bonding layer for the other interface layers.

Hausmeister486, 2024-07-07:
> UP! Please, it would be awesome! Actually it‘s useless!!!
The current multi-material possibilities are clearly behind the hardware in terms of software.
Unfortunately, the Bambu Studio is slowly outperforming the Prusaslicer.

Hausmeister486, 2024-07-19:
Reminder:
I still have hope, I'm not giving up yet.

Please make the Prusa XL the best multi-material printer on the market

morphias2004, 2024-07-20:
It is so easy for them to implement this as it is already available in derivative slicers. All they have to do is merge the changes.

This issue combined with #12688 are critical for excellent support interface layer success.

morphias2004, 2024-08-16:
Still an issue.

argosforge, 2024-09-17:
Still a problem

Hausmeister486, 2024-11-30:
Still a problem

Hausmeister486, 2024-12-24:
Still a problem

Hausmeister486, 2024-12-28:
Suggested solution for the Prusa Slicer:
Flexible selection and settings of support structures and interface layers in terms of the number of layers and which material or extruder is used for each layer.
The whole thing separately for the top and bottom of the support.
A global bed temperature setting even when using different temperatures of the support materials.

npburns224, 2025-04-09:
I stumbled across this issue after spending the day tweaking my support settings without success. It seems impossible to print multi-material support interfaces reliably given the current implementation. I wasn’t previously aware of how Bambu Studio handles printing the interface layer in another material, but it seems like had the right idea.

I hope Prusa implements something similar soon as it’d be a _huge_ improvement. Until then, I guess I’ll resort to modeling the “interface layer” in CAD.

morphias2004, 2025-07-21:
Over a year on and this still hasn't been implemented. :(

Hausmeister486, 2026-09-18:
Over two years on and this still hasn't been implemented. :(

morphias2004, 2026-09-18:
> Over two years on and this still hasn't been implemented. :(

Absolute joke hey?!

I love how they've just closed all the open cases because PS3 is coming out. Initial testing shows most of these issues have not been addressed.