prusa3d/PrusaSlicer #12063: Protruding outter perimeters

gaaZolee, 2024-01-15:
### Description of the bug

I was printing a scale up  (4x each direction) 2 color Benchy on my Prusa XL. FInal result was very satisfying, however it had on glitch. Print had ~2 outer perimeters which are protruding out of the desired shape. It protrudes symmetrically on both sides and disappearing close to the nose. I investigated it a little. It looks like that the glitch is directly where the "floor starts" See the included project and image. 

Actually I was not sure if it is a Printer  firmware or Slicer issue so I am reporting it here.

Print setting was 0.32 layer height and speed wit IS and 0.56 mm nozzle.
Used filaments PLA. Prusament Galaxy black and Polymaker Wine Red

[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

Project file: [Snapshot from slicer with first suspected layer - Big-Benchy_2PLA_XL.zip](https://github.com/prusa3d/PrusaSlicer/files/13933938/Big-Benchy_2PLA_XL.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

Prusa XL multi tool (5)

rtyr, 2024-01-15:
See https://help.prusa3d.com/article/the-benchy-hull-line_124745

There are also duplicate issues on this topic, so I am closing this.

closed 2024-01-15 by rtyr

gaaZolee, 2024-01-16:
Ohh thanks. I didn't see it before ...