prusa3d/PrusaSlicer #12066: Top layers not generating on spherical bowl

cromine11, 2024-01-15:
### Description of the bug

When slicing, the top layer don't generate on a few layers and it only applies bridging infill multiple times

### Project file & How to reproduce

[Spherical ear cup.zip](https://github.com/prusa3d/PrusaSlicer/files/13938791/Spherical.ear.cup.zip)
I haven't done anything unusual. the only thing that affects it is increasing top layer count
[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

windows 11

### Printer model

Voron Trident

mix579, 2024-01-15:
Can you upload the 3mf project file with your settings? I loaded the STL into Prusaslicer with standard 0.2mm profile for Mk4 and don't see what you're seeing.

cromine11, 2024-01-16:
That is a 3mf with the settings causing it. It checked it some more and the setting that affects it is infill density at 40% or higher. lowering the number of top layers will make it more visible.

closed 2026-09-29 by prusaslicer-bot[bot]