prusa3d/PrusaSlicer #12072: Keeps saying Relative extrusion requires RESET when it is already ON

inspectionsbybob, 2024-01-16:
### Description of the bug

I am trying to use a printer that uses Relative extrusions.

My Printer Settings General already has "Use Relative E Distances" checked...

I open the Slicer and check the Printer Settings; Relative E is checked.

I Import/drag-n-drop a STL onto the plate
I immediately get:

**ERROR
Relative extruder addressing requires resetting**

I tried saving it as a 3MF (project) and then opening it...

Same error

### Project file & How to reproduce

Open the 3MF file, and you will get the error immediately
[Aquila_Camera_mount.zip](https://github.com/prusa3d/PrusaSlicer/files/13944231/Aquila_Camera_mount.zip)
[image]
immediately

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Win11

### Printer model

Tronxy X5SA

sarusani, 2024-01-16:
Click on "More" and follow the instructions. :)

In short:
Go to Printer Settings -> Custom G-code -> Before layer change G-code
Enter:
```
;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
```

That should fix it.

closed 2024-01-16 by rtyr

inspectionsbybob, 2024-01-16:
Ok this is new, never had this issue on earlier releases.  Will this be fixed in future releases?  I will try this when I get another similar print.

sarusani, 2024-01-17:
When you enable "Use Relative E Distances" you need to reset the position between layers. It has always been like this.