prusa3d/PrusaSlicer #12074: Missing layer

Jaggana69, 2024-01-16:
### Description of the bug

Hello,
The walls of my object are full, except a few holes, after slicing, there are entire layers missing on one wall
[image]
[image]

### Project file & How to reproduce

Slice the file with the settings
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/13951953/files.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+

### Operating system

Windows 10

### Printer model

Wanhao D12 300 PRO MAX

neophyl, 2024-01-16:
Hi.  First of all, you only need to attach the Prusa Slicer project file.  The 3mf already contains the model as well as all the settings you are using.  No need for the ini etc 

When I open your prusa slicer project file in the right hand tree list you can see the yellow warning triangle.  Please pay attention to this as it is telling you that something is wrong with your model.
Hovering over it we can see that Prusa Slicer has already repaired 1630 errors but that there are still 432 open edges.  It is these that are causing the model to slice as it does.
[image]

If you are on Windows you can click the triangle and it will use a built in api available to windows to attempt to repair the other errors.  If you are on a different OS then you will have to find an alternate way to repair the file.

The Windows api does successfully repair the file though.  
[image]

I see that you mention you are on Windows so please use the windows fix by clicking on the triangle.

Jan-Soustruznik, 2024-01-17:
Hello, @Jaggana69, 

according to @neophyl comment above, your model is corrupted, when model is repaired, then problem is solved. 

I just add link to Prusa knowledge base for more info: https://help.prusa3d.com/article/corrupted-3d-models-for-printing_2205 

Closed as solved.

Jaggana69, 2024-01-17:
Hello @neophyl and @Jan-Soustruznik,

Thank you for your fast replies.
I just exported my Fusion360 model to an STL and I didn't knew it can be corrupted/broken! :O 
So Fusion360 is badly exporting STL files? Is it a known issue?

Indeed, after right clic + "Fix by Windows repair algorythm" there are no more failure after slicing.
Thanks!