prusa3d/PrusaSlicer #12075: No way to limit Extrusion width for a given Material/Extruder, Make arachne super hazardous to use

ProTubeVR, 2024-01-16:
### Description of the bug

The problem is known, but all requests are 'closed',
(#9150 , #12026 ....)
To me, this problem is so painful that we can't consider the Arachne as functional without a way to limit maximal extrusion width.
 At this moment, the width can be 2x larger than the the dialed setting, how you can imagine it could go well ?
 
[image]!

I'm using 0.5mm nozels, so would be great to limit this to 0.7~0.75mm... this would make arachne usable, cause atm it's super Hazardous feature that user can't control. 

Ps : this isn't a feature request, but a Fix request for a bugged feature 

### Project file & How to reproduce

just slice anything using Arachne, and you'll see many 'fat' lines, and no setting can solve it

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

MK3s, MK4, XL

Tupson444, 2024-01-20:
I'm not against your suggestion, but I don't see a problem with wide extrusion width (I may be missing something though). 2x nozzle diameter or even more is perfectly doable, I use it often with high flow nozzle on a slow printer (to max out flow without fast XY movement).

ProTubeVR, 2024-01-24:
> I'm not against your suggestion, but I don't see a problem with wide extrusion width (I may be missing something though). 2x nozzle diameter or even more is perfectly doable, I use it often with high flow nozzle on a slow printer (to max out flow without fast XY movement).

this isn't possible to handle properly depending on the material you use,
To get 'normal' perimeters on my case (not under-extruded ones), i would need to limit 'over width' for a given Material or nozel/extruder.

To me, this is a base setting to have 'control' on the slice,
As it's, it's super hazardous to slice in arachne.

We have a obliviously a minimal element setting, cause we need it,
We should have a maximal element setting, cause the extruder have physical limitation ---> so we need to take this limit under consideration or it fail to get a 'clean perimeter' printed.

krushia, 2024-08-29:
Generally the extrusion width should be less than the total surface area of the tip of the nozzle. I currently use a 0.4 mm nozzle that only has 0.6 mm of total surface area (the stock Voxelab Aquila nozzle). There is no place for me to specify this parameter in the extruder settings.

I realize the Arachne widths can be somewhat controlled using other settings, but those are located in the "Print settings" instead of "Printer settings" and also unintuitive (see #9150).

closed 2026-09-29 by prusaslicer-bot[bot]