prusa3d/PrusaSlicer #12077: [FR] Multiple dovetails / pattern / multiple cutting panes with dovetails on a cutting line

lexagigabit, 2024-01-16:
**Is your feature request related to a problem? Please describe.**
Sometimes I need more dovetails than one, for a single solid body, or for a body with space in between. Please see attached image for an example.

**Describe the solution you'd like**
I would like to have multiple dovetails or a pattern with the adjustable distance on a single body / on a single body with spaces

**Describe how it would work**
Either/or:
- creating multiple cutting panes on a single cutting line so that each dovetail might have its own parameters
- creating multiple dovetail joints on a cutting pane, with the adjustable distance between each one
- allow defining a pattern, i.e. 'insert a dovetail each X mm starting from the first one', then when the first dovetail is ready, briefly show the location of the second, third, etc. Allow changing the distance from the same window.
Important. Multiple dovetails should be allowed on a body with spaces too. Please see attached image as a reference.

**Describe alternatives you've considered**
An alternative is creating dovetails in CAD. Unfortunately that's not always possible, especially when the sources are not available. Also it's time consuming.

**Additional context**
The image attached has a frame, which, as this FR states, could be cut in half left-to-right with 2 dovetails.
First dovetail (left side) is visible and ready to be used; Second dovetail (right side), unfortunately, is not doable in the current version.
[image]

rajdude, 2024-01-17:
It's funny that I just spent an hour looking for the same feature!
I want to 3dp a large flat plate like object. I want to cut it but have multiple dovetails at the cut line, so that the two parts can be snapped together.

adenta, 2024-01-18:
Also interested in this!

thebige3, 2024-01-27:
Yes, the ability to add multiple dovetails is something I'm looking for also.

rajdude, 2024-03-11:
Well, I spent a huge amount of time on this. Frankly I have given up. I designed dovetails manually in Inventor. I found out that it is very difficult to come to a size which actually snaps together nicely. The 3D printer does not print these things super accurately, in terms of dimensions. 

Then later on I found out that 3D printed parts are glued together very easily. So now I just design my parts with a simple "lap joint" and glue them.

lvisintini, 2024-03-11:
I found a way to do this on Freecad and I can totally see that it would be quite a challenge to come up with a reasonably intuitive UI that would allow you to do this

Matrixrs, 2025-07-31:
I would like it to !

Honza51, 2026-09-02:
Please keep this issue in mind. In version 3.x.x, it is still limited to a single dovetail. This issue is still relevant, and a lot of users (Prusa fans) would like to see the option for multiple dovetails. It would be a great feature to have.

Patronics, 2026-09-02:
Yep, this would still be a very useful feature in 3.x! Perhaps it's something the plugin system could eventually allow too, if not officially implemented before then