prusa3d/PrusaSlicer #12085: G1 F0 error when supports enabled

stoney7713, 2024-01-17:
### Description of the bug

I keep getting the G1 F0 error when enabling "supports on build plate only" and  Top interface layers "1 light" causing an error and prints will not continue.

If I set "supports everywhere"  or change Top interface layer to "2 default" both resolve the problem.

### Project file & How to reproduce

Problem file
[Thuan filter cup2.zip](https://github.com/prusa3d/PrusaSlicer/files/13969869/Thuan.filter.cup2.zip)

Solved setting support to "supports everywhere"
[Thuan filter cup2fix.zip](https://github.com/prusa3d/PrusaSlicer/files/13969872/Thuan.filter.cup2fix.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11

### Printer model

Creality Ender 5 S1

khumarahn, 2024-03-10:
I have a lot of `G1 F0` without any supports

khumarahn, 2024-03-10:
The `G1 F0` bug is sensitive. For me it was enough to move slightly the parts on the plate, and it disappeared....

closed 2026-09-29 by prusaslicer-bot[bot]