prusa3d/PrusaSlicer #12089: Strange behavior arachne perimeter generator

aclucca, 2024-01-18:
### Description of the bug

Wall thickness is form bottom to top 1mm.
Arachne perimeter generator calculates sometines 1 perimeter and sometimes 2 . . .

### Project file & How to reproduce

Slice with standard 0.33 speed setting
[arachne.zip](https://github.com/prusa3d/PrusaSlicer/files/13981479/arachne.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10 64bit

### Printer model

PRUSA MK4

Martin-Danger, 2024-01-19:
Interestingly, reducing the value for Advanced - Slicing - G-code Resolution to 0.0059 removes the occasional single perimeter issue. Not a fix of course, but might point to the cause of the issue.

neophyl, 2024-01-19:
Looks like an edge case.  The thickness on the curved corners varies slightly from place to place as the surface is built up of triangles so some places are for example 1.0031, others are 1.0095 just to pick 2 examples.  The corners are 'split' into 3 vertical sections and those different layers are around the boundaries.
[image]

Then with Arachne the configured extrusion width it is sometimes deciding it can fit that perimeter in as a single 'thick' one and others as 2 'thin' ones.  
Change the extrusion width for external perimeters to 0.65 and it doesn't do it (edge case gone).  Or reducing the resolution also changes the occurrence of the edge case due to the math.
You can also stop it by changing the arachne minimum perimeter width to 87% which removes the math edge case.  
Probably other changes to the settings will remove the edge case too.

If you modify the model to remove the 3 vertical sections 
[image]

and then slice with your original settings you get 
[image]
where it decides it can fit every perimeter as a 'thick' one.

So I don't think this is a bug, just an unfortunate concentration of circumstances with the model construction and configuration.

aclucca, 2024-01-19:
Thank you for the research and explanation. 

This means that a width change sub 1% (or sub 10µm) will cause this effect? I think to prefent this, the calculation should implement a hysterese or something like this . . .

aclucca, 2024-01-19:
[image]
[image]
[circle.zip](https://github.com/prusa3d/PrusaSlicer/files/13994082/circle.zip)

I´ve uploaded an example designed in Fusion 360 and exported in "High" Setting to prusasclicer.
The mesh structure exported from fusion dont correlate with the slicing result of prusasclicer . . .

closed 2026-09-29 by prusaslicer-bot[bot]