prusa3d/PrusaSlicer #12099: Saving project or Closing program throws Windows-error "PrusaSlicer doesn't work anymore" ("PrusaSlicer funktioniert nicht mehr")

schleifi, 2024-01-20:
### Description of the bug

When there is a sliced model, the program cannot be properly closed and the error message (translated from German):
"PrusaSlicer doesn't work anymore". The same happes when a a project should be saved.
The error does not happen when the model is not sliced, also if it is sliced and then a parameter is changed and the model is not sliced anymore.

### Project file & How to reproduce

cf bug description

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

win10

### Printer model

mini+

Jan-Soustruznik, 2024-02-01:
Hello, @schleifi , I cannot reproduce reported problem. Could you provide more info, please ...  model, 3MF file, HW spec, error logs that may help to investigate your problem. 

Thank you, have a nice day

schleifi, 2024-02-01:
Hi @Jan-Soustruznik ,
I get the behaviour independent of the STL:
import STL, slice with Input Shaper profile (0.2mm Speed; no changes), export bgcode to USB drive, eject USB-drive from Prusaslicer, close PrusaSlicer or "save project as..." -> error "PusaSlicer not working"
This has been essentially my constant workflow and reproduces reliably the error on my computer.
(However I also got an error under different conditions before posting the error-report here; since yesterday I got a new driver for the graphic card and appears to be more stable - no clue whether there is any correlation or just pure luck and present testing is different from before).

Below is the error-log from Windows.
If you need more information, let me know

Beschreibung
Pfad der fehlerhaften Anwendung:	C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe

Problemsignatur
Problemereignisame:	APPCRASH
Anwendungsname:	prusa-slicer.exe
Anwendungsversion:	2.7.1.0
Anwendungszeitstempel:	657872dd
Fehlermodulname:	PrusaSlicer.dll
Fehlermodulversion:	0.0.0.0
Fehlermodulzeitstempel:	657872c8
Ausnahmecode:	c0000091
Ausnahmeoffset:	0000000000585ad5
Betriebsystemversion:	10.0.19045.2.0.0.768.101
Gebietsschema-ID:	1031
Zusatzinformation 1:	6644
Zusatzinformation 2:	66443637b52422d87eef4e39743cadba
Zusatzinformation 3:	c380
Zusatzinformation 4:	c380a34f6ee9f0d96db7fa7f0fb63475

Weitere Informationen über das Problem
Bucket-ID:	b1256a033684ed7229214971b1c55ce0 (1810809277842283744)

closed 2026-09-29 by prusaslicer-bot[bot]