prusa3d/PrusaSlicer #12101: Top Layer Gap

ALD-Models, 2024-01-20:
### Description of the bug

The top layer of the print some is solid filled however for some of the top layer their is a 0.45mm gap 

[image]

### Project file & How to reproduce

[https://1drv.ms/u/s!Ag3vv7s3QoSihbRhLkjm2HRI_cpjEA?e=sjexVI](url)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.0

### Operating system

Windows 10

### Printer model

Prusa Mini+

Jan-Soustruznik, 2024-01-31:
Hello, @ALD-Models , could you re-share your project file? I cannot open attached link. Thank you, Have a nice day

ALD-Models, 2024-01-31:
This should work:
`https://drive.google.com/file/d/1zz8XO4oLqbVRTgSZfAIfLPDZMst02DXj/view?usp=sharing`

closed 2026-09-29 by prusaslicer-bot[bot]