prusa3d/PrusaSlicer #12107: Organic Supports: 'Layer Shift'?

cmuche, 2024-01-21:
### Description of the bug

I experienced a weird problem when using organis supports which look like a layer shift. It seems like the tree is shifted from one layer to another.

This caused a failed print since I didn't mention this before and because of the mid-air-printing the print curled upwards at this point.

I can't find out if there is a particular setting causing this problem. Tested with v2.7.1 and v2.7.0.

[image]

[image]

[image]

[image]
[image]

[image]
[image]

### Project file & How to reproduce

[Giraffe -Version 1 - NORMAL 0.20mm - ERYONE Matte PLA 195C.zip](https://github.com/prusa3d/PrusaSlicer/files/14002670/Giraffe.-Version.1.-.NORMAL.0.20mm.-.ERYONE.Matte.PLA.195C.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

Ender 3 V2 Neo

neophyl, 2024-01-21:
Duplicate of issue https://github.com/prusa3d/PrusaSlicer/issues/10725  (and I think several closed ones)

closed 2024-01-21 by rtyr

xrbeattx, 2024-05-02:
Why are all of these closed? This issue is still happening in 2.7.4?

rtyr, 2024-05-02:
Only duplicates are closed.