prusa3d/PrusaSlicer #12111: temperature resets to old values when setting a new filament profile and then save this new profile with its own (new) name.

3dcase, 2024-01-22:
### Description of the bug

I made a new filament profile which I had not used before, Spectrum ASA 275. I changed the settings for the temperatures for nozzle and bed and saved this profile with the new name Spectrum ASA 275. I make a few more settings and slice my model. Then I print it and I notice that the temperatures in the gcode are not what I saved in the profile. I check my slicer and the temperatures had reset themselves back to the previous filament I used BUT the profile is named the new filament Spectrum ASA 275.
This has happened before to me but because it was the first time I thought I had done something wrong BUT I was almost sure the slicer made this change. On changing it for the second time the new values remain and it works as expected.
The project file is after I changed it for the second time, so they show the correct values.
The screenshots are sequential and show what I did, saved and then what became of it after I made more changes.
First screenshot is when I changed it , second screenshot is after I saved it, then made other changes without saving again and went back to the temperature screen to check. The values are the old ones again!

### Project file & How to reproduce

[universal_machine_key.zip](https://github.com/prusa3d/PrusaSlicer/files/14013096/universal_machine_key.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Linux Ubuntu 22.04.3 LTS

### Printer model

CR10S PRO V2 and CR10 SMART PRO. Both machines had this happen

Jan-Soustruznik, 2024-01-30:
Hello, @3dcase , 

I cannot reproduce your reported problem. Could you provide more info, please? This problem starts in 2.7.1 or you can reproduce in older version?

Thank you, have a nice day

3dcase, 2024-01-30:
Hello Jan,
I don't know about the old version, it never happened before I switched to 2.7.1 and I am not using the old version anymore.
All I did, and it has happened every time now since I noticed and I pay attention to it, is that when I want to make a new profile and I change temperature of the filament or bed, when I save it with a new name and I continue to change other parameters, when I check back after I made other changes, the old values have reappeared in my first changes. Yesterday I made a new profile for a nylon filament after I printed a temp tower from teaching tech, And I change speeds as well as temperatures. This time it changed the speeds back to the previous values and I had to re-enter them again. After the first time they change and you re-enter them correctly, they stay correct after that. It is only the first time after you save the new profile with different parameters like temperature or print speeds that when you change other settings after the first save, and you check back that the first settings you changed have reverted.
I am sorry I cannot give you more details and that I cannot use my old version anymore. If you tell me how I can experimentally use the old version again I could try to mimic the whole thing there too for you.
Best regards and hope you find something useful in my description above.
Kees (3dcase)

Sent with Proton Mail secure email.

lukasmatena, 2024-02-12:
> Any news yet or is the update for 2.7.2 beta version addressing this issue?

We did not manage to reproduce it. If you want to try an older version (e.g. 2.6.1), just download an AppImage from [our github releases page](https://github.com/prusa3d/PrusaSlicer/releases) and run it. By the way, where did you get the binary? Is it from our github, or from some Linux source like package manager, flatpak or something like that?

lukasmatena, 2024-02-12:
> I try not to use these linux packages as they often do not work at all or badly.

Agreed - that's why I was suspicious.

You can also include a short screencast (ideally from the point when you start PrusaSlicer). There might be some little detail which triggers the problem and which we fail to reproduce.

closed 2026-09-29 by prusaslicer-bot[bot]