prusa3d/PrusaSlicer #12114: Add section view to slicer

MitchRex723, 2024-01-22:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Cant control supports inside cavities in model
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
A section view of the model
**Describe how it would work**
Be as descriptive as your skill level allows you.
Click a section view button. Can switch between a section on the x y or z plane. Can move the plane to get it positioned.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I have tried using support blockers but does not work with internal cavities
**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2024-01-23:
That function is currently done using the clipping view inside the support painting tool. It allows you to visually remove part of the model from the angle the view is at to see inside a model and paint on support enforcers or blockers. 
It’s a little clunky to use sometimes but does work. 
Please see the knowledge base article https://help.prusa3d.com/article/paint-on-supports_168584. As it has a demonstration of using the clipping view.

MitchRex723, 2024-01-23:
Works great thank you

Cdddo, 2024-01-27:
> That function is currently done using the clipping view inside the support painting tool. It allows you to visually remove part of the model from the angle the view is at to see inside a model and [...]

It is really not useful if its only in the painting mode. The painting mode also per object and hides all other objects, which makes it useless when you want to position parts or modifiers.

neophyl, 2024-01-27:
You are basically talking about a different function.  
I opened a feature request in Feb 2020 for the exact thing you want https://github.com/prusa3d/PrusaSlicer/issues/3678  a similar one was opened later with a better description of the possible changes https://github.com/prusa3d/PrusaSlicer/issues/4438

kubispe1, 2024-10-29:
I am closing this issue. The clipping view is already implemented, and the idea around a transparent or wireframe view that would help with positioning modifiers and volumes inside larger objects is described in #3678 and #13534.

closed 2024-10-29 by kubispe1