prusa3d/PrusaSlicer #12117: A GL anti-aliasing filter for the slicer viewer

amirshachar, 2024-01-23:
The GL viewer simmers and struggles to display fine lines when viewing the slicing result, especially when slicing a more complicated model.

Similar to how DCCs display wireframes and fine lines in real-time renders, the GL viewer in slicer can/should employ similar anti-aliasing filtering technique for a cleaner more stable render view of the sliced model

It would not need any user input but rather just filter the rendered result in the view

[image]

katp-hyphen, 2026-09-02:
I would like to attract some attention to this issue now that 3.0 is coming out. Since many other graphical effects got implemented, aliasing now remains the most obvious graphical issue with the 3D view. I think the shadows might actually be making it worse. 

[image]

katp-hyphen, 2026-09-02:
This applies to PS3.0 too.