prusa3d/PrusaSlicer #12119: Feature request: changing the slicing algorithm to reduce warping (of ABS parts) in FDM printers

ag88, 2024-01-23:
**Is your feature request related to a problem? Please describe.**

hi, this is a feature request.

Could prusa consider some change in the slicing algorithms to reduce warping of long (ABS) parts in FDM printing?

In my google searches around for solutions to reduce warping for long (ABS) parts. I stumbled into the following papers:

[Warpage of FDM parts: Experimental tests and analytic model](https://re.public.polimi.it/retrieve/e0c31c0f-8530-4599-e053-1705fe0aef77/RCIM-2018%20postprint%20%281%29.pdf)

in the above paper / link, it shows that the continuous length of parts is a key factor that is observed in warping, lifting of the prints. I think the paper is well researched and coincides with what I casually observed as well, longer parts warp more severely and requires a lot of counter measures, use a heated bed, turning off part cooling fan, use an enclosure, etc, but it didn't fully resolve the issue. A heated bed probably helps with the 1st few layers, but cracks form at layers beyond the 1st few layers.

**Describe the solution you'd like / Describe how it would work**
Be as descriptive as your skill level allows you.

Next I stumbled into this paper

[Bricking: a new slicing method to reduce warping](https://www2.uned.es/egi/publicaciones/congresos/Bricking_A_new_slicing_method_to_reduce_warping.pdf)

It turns out hexagonal shapes bricks help to reduce warping ! and the idea is to breakout long stretches into bricks so that the continuous length is shorter in each direction and hence reduce warping. The bricks can probably be interleaved so that the edges don't all fall along the same edge, e.g. as like brick laying in masonry buildings.

Could this be looked into and perhaps start with experimental features so that Prusa slicer users can try them out?

This can perhaps be an optional selection as like infill etc, just that this is different from infill in a sense that it affects the boundaries as well.

This has potential to help solve warping in large flat layer prints such as printing box enclosures, which often has large lengths by virtue of the function needs.

 Thanks for reading,
  Cheers

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

mentioned prior above

**Additional context**
Add any other context or paint drawing about the feature request here.

Take a look at the paper in the solution section, the idea is to break long continuous lines into *bricks*, so that it limits the warping / lifting for each short brick. The bricks can probably be interleaved so that the edges don't all fall along the same edge, e.g. as like brick laying in masonry buildings.

murk-sy, 2024-01-25:
I've looked through the paper, and while it sounds interesting, I feel like the most important information is missing or was not accounted for, and that is time spent per layer.

You can print ABS very slowly to combat warping. Their listed methods showed a significant reduction of warping (9% at lowest vs 35%), but unless you are comparing parts with same layer time (and ideally same or similar strength), it's not really an apples to apples comparison. If their method makes the print take twice the time, is the result actually better than just printing at half the regular speed?

If you create some test models (already cut up to blocks as noted in the paper) to actually compare this it'll be a lot easier for someone to focus some time on it.

While writing this I got some ideas of doing something similar by either using slits in long straight sections of models (like pipe expansion joints) or significantly slowing down the print when laying down longer sections - either fully or having short sections of slowdowns on perimeter every x mm. I might make some examples if time permits.

Tupson444, 2024-01-25:
Interesting direction of thinking. If someone finds a way to use something like this, it would be great to have it automatically done in the slicer.

>slits in long straight sections of models

I also tried using slits in long sides of a model (by placing tall and thin negative modifiers to "cut" the perimeters in multiple parts), in hope it will help reduce warping when printing ABS. However I don't know how much it helped (if it influenced anything at all), as I haven't done any thorough testing.

ag88, 2024-01-26:
At the moment, these 'cuts' can only be done in cad to reduce the continuous run length, but that it'd be good to have a slicing algorithm to do this as the paths that can be traced goes beyond what can be defined in cad and further optimization of the tool path to avoid spurious movements due to the rather complex geometries splitting things into 'bricks'

vovodroid, 2024-02-04:
This  **Reverse extrusion directions #12126** PR also address warping issue.

kubispe1, 2024-04-19:
Hi, @ag88
I've been thinking about "bricking" concept for a while now, and I'm sure it could work however, there are some fundamental challenges. This simmilar logic is used for example:
- to reduces concrete shrinkage.
- welding massive joints to minimize deflection of the final parts.
challenges I see:
- Dividing infill and perimeters into "bricks" (sub-parts) might compromise the overall strength of the model.
- Dividing perimeters could affect the aesthetics due to the introduction of new seams.
- The additional travel paths increased printing time and could produce other issues.
[image]
(extracted from paper: Bricking: a new slicing method to reduce warping, 2nd one)

ag88, 2024-04-19:
[kubispe1](https://github.com/kubispe1) thanks for your response, I'd guess this 'bricking' which reduce warping for long abs parts works, and it'd be applicable for more 'specialized' models, i.e. those that can tolerate 'holes' between the layers in the walls etc. for the time being I can achieve 'bricking' in the stl model e.g. to 'poke holes' say on the walls of a casing for example. this is easy say in openscad and perhaps some parametric cad, but could be a challenge in graphical cad.

either way this is just a feature request / idea, this request can be closed if it is deemed not ready to be implemented.
it may be considered in future etc.

vovodroid, 2024-04-22:
> Dividing perimeters could affect the aesthetics due to the introduction of new seams.

External perimeter could remain continuous.

dbaarda, 2024-11-05:
Just stumbled on this while looking for anything that implements https://www.cnckitchen.com/blog/brick-layers-make-3d-prints-stronger.

This is a bit of a different thing; this bug and referenced papers are about rendering layers as "bricked polygons", whereas the link I added is using brick-laying to give "interlocking layers".

AFAIKT "bricked polygons" is a bit like putting thermal expansions gaps in the layers. which would weaken the overall strength and give you visible artefacts (that could maybe be mitigated by not "bricking" the outer layer). It still might be worth it to reduce warping.

The "interlocking layers" idea is to boost layer-bonding strength and not specifically to reduce warping, but I'm betting it could also help reduce some warping artefacts because;

1. The increased layer bonding strength would reduce delamination from warping, which would help hold the piece together and straight.
2. The alternating "stripes" on each layer are not bonded to each other, they are bonded to the "other stripe" on the layer below. This means they bond to a "cooled" stripe that has already thermally shrunk and stabilised, instead of the whole surface undergoing thermal shrinkage at once for each layer.

I suspect this would mainly reduce warping artefacts "across the grain" of the extruder-path on the layers, but I bet it would also reduce longitudinal warping artefacts because of of reason 2 above.

If this works, it would both reduce warping artefacts *and* increase layer-bonding strength.

The only minor issue is the outer top surface would have a castellated grain texture unless you finished off the surface with a half-height bricked-layer to fill the gaps, or maybe iron it down quarter of a layer height.

Tupson444, 2026-09-18:
still relevant