prusa3d/PrusaSlicer #12124: Microsoft Repair tool SUCKS!!! It fills in intentional voids I've included for magnets etc.  It took me HOURS and multiple failed prints to figure this out!

hackerboysf, 2024-01-25:
### Description of the bug

When I generate a design in Fusion 360, the Prusa Slicer now tells me there are errors in the design. When I used to see this flag (rarely, and always with STL files I'd generated using online services) the "fix" used to be some online service that I cannot remember.  Now, even with designs I've generated myself that have nothing fancy on them, and which I KNOW my printer is capable of printing without a problem, the Slicer indicates that there's a problem, and the only option presented is the Microsoft repair tool.  In the design I'm printing now there are multiple voids that will hold magnets to keep things together, but the first several prints didn't have the voids that were imported into the Prusa Slicer.  They were erased by the Oh So Brilliant Microsoft Repair Tool.  They are arrogant and think they're brilliant, but they are sloppy.  This cost me many printing hours to figure out-  a waste of time, filament, and attention.  
Also, if you're dealing with 3D printing issues, why does this GitHub upload limit the file types to image file types?  I'm trying to send the before and after STL files and I can't upload them... just image files.  

[image]
[image]

### Project file & How to reproduce

[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 10

### Printer model

Lulzbot Taz 6

neophyl, 2024-01-25:
You need to zip up files.

sarusani, 2024-01-25:
You don't have to use the MS repair tool. Feel free to use whatever you prefer.

stoxx, 2025-12-03:
Sounds like an intentional function of the repair algorithm and it's possible Prusa cannot help it much. 

But, out of curiosity: Did you try if adding a microscopic hole that connects the void with surface would help the repair tool to realize not to remove the void? 

It would create some extra walls when printing but those are likely less harmful and improve model strength locally. Depends. 

But an interesting find in any case, thanks for the heads up.

closed 2026-09-29 by prusaslicer-bot[bot]