prusa3d/PrusaSlicer #12132: (REQUEST) Half Z-HOP

rhlacerda, 2024-01-26:
Hi! I have a K1, and I am printing some small models. I need to use Z-hop because my models have some small details, and when the hotend travels... sometimes hit and break the model.

I tried all types of hops and it leaves strings or some blobs on it. Lost a lot of time trying to configure retraction, temp and other stuffs to solve it... but when I turn off z-hop... I got a perfect and clean print... but I can't use it off, because I want to print 30 models at the same gcode, if the nozzle hit a model... GG gonna be a mess.

So I got one idea... What if we have a half z-hop process...
Will work this way:

1- the nozzle leave the model like z-hop is disabled;
2- moves up when the nozzle isn't over the print;

something like this:
                     _______
_____________/           \____
MODEL                    MODEL

a mix of off z-hop + slope z-hop

I am using wipe with z-hop on and z-hop models. Tried without it too... but the best print has wipe without Z-hop.

Thanks! Hope it is possible and can help every maker =)

smurfix, 2024-03-28:
NB when you use ASCII graphics here, it's a good idea to mark them as pre-formatted, i.e. enclose them with lines consisting of triple backquotes.

neophyl, 2024-03-28:
Isn't this just ramping lift ?  Settings Under Printer Settings>Extruder.

closed 2026-09-29 by prusaslicer-bot[bot]