prusa3d/PrusaSlicer #12136: Slicer messing up single line font stroke width

ronperek, 2024-01-27:
### Description of the bug

When it comes to printing multicolor filament, single line text with a font that has constant stroke width, the slicer seems to have a mind of his own making some of the lines wider. This negatively affects text print quality.
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[font test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14070216/font.test.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.1+flathub.org

### Operating system

Linux Mint 21.3 Cinnamon

### Printer model

MK3S

Jony01, 2024-01-31:
Could you provide font file?
(from 3mf I see, you are using font with name `print-ron`)

ronperek, 2024-01-31:
[print-ron.ttf.zip](https://github.com/prusa3d/PrusaSlicer/files/14110408/print-ron.ttf.zip)

ronperek, 2024-02-12:
Did you manage to replicate the issue?
Maybe you can look also at the issue #12135. It also applies to single line font printing.

Jony01, 2024-02-12:
Yes, I replicate it. I can clearly confirm the issue is there.

I found, that this is not connected with text emboss feature.
It is caused by arachne. We collect the issues and will investigate it further later. 

At this moment, I can not provide you a solution for you. Sorry.

ronperek, 2024-06-09:
Hi there! Just wanted to follow up on this issue. A few months had passed, and I’m still experiencing the problem in version 2.7.4. This problem contributes to lowering the quality of print. It makes font characters ugly. Especially the '+' sign. I had to scrap a few kg of the prints because the quality was just too bad for the end product. Partially it was because of this issue. Is there any additional information I can provide to help resolve this? Thank you for your time and effort!
[image]

ronperek, 2024-12-12:
Hello. Just want to report that the issue is still present in 2.8.1.

Jony01, 2024-12-13:
Sorry, currently there is no deep investigation connected with particular this issue.
Have you tried to:
* Not use arachne for one line thick text?
* increase size OR boldness of glyph?
* use another font?
* Go around by yourself? 
(Which dead end you found?)

Priority of PrusaSlicer development are at this moment elsewhere.

ronperek, 2024-12-13:
Not use arachne for one line thick text?
Classic skips single line extrusions. Even some are skipped when "Detect thin walls" option is checked. Generally it's worse.

increase size OR boldness of glyph?
Not an option. I don't have space for size or boldness increase.

use another font?
I already made and optimized this font myself to make sure all the extrusions will be single line, which gives the best quality and reliability over other publicly available fonts.

Go around by yourself?
What do you have in mind? Is there some workaround to disable the slicer adjusting the extrusion width this way?

ronperek, 2024-12-29:
Settings:
[image]

Before slicing:
[image]

Arachne:
[image]

Classic:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]