prusa3d/PrusaSlicer #12144: Add an option for setting bridge/perimiters overlap

cerja, 2024-01-27:
There are still unsolved issues with bridges not connecting to the perimeters (#11500 - open, #10231 - closed but unresolved #9988 - closed as of 2.6.0-beta1, but the problem apparently persists).

The problem is most likely a combination of slicer algorithm and printer specific  traits which result in either unconnected or poorly connected bridges. As for now the issue appears to have been tackled mostly by increasing the **Infill/perimeters overlap** which, however, invariably starts causing issues elsewhere on the print.

Thus I propose an addition of a second parameter named **Bridge/perimeter overlap** which would handle bridging overlap separately from other infill.

VooDisss, 2024-01-29:
I have did quite many tests (about 60 prints of bridges with various bridge speeds(10-140mms), flow ratios(0.7-1.2), bridge accelerations(50-150mms)) Trying to get the perfect settings for bridging... So it seems this was the problem!

I found decreasing the bridge acceleration to 50-150 interval helped the most somewhat, but still these bridges are not useable.
Going to revert back to 2.5.0 or older version, where this issue does not occur (as mentioned at https://github.com/prusa3d/PrusaSlicer/issues/10231#issuecomment-1553310739)

brandonbeaumia, 2024-03-12:
Add #9999 to the list (I just added screenshots and a project file to that issue); we're now at the one-year anniversary of the first report.

l29ah, 2024-06-11:
Still occurs as of version_2.7.5-rc1

Tupson444, 2026-09-18:
close only if resolved