prusa3d/PrusaSlicer #12145: Dynamic Overhangs OVERRIDE?? Max. Filament volumetric speed

total1ty, 2024-01-28:
### Description of the bug

When printing speed is limited by the maximum volumetric througput set in the filament profile, the dynamic overhang speeds seem to ignore this and move faster than the filament profile would allow.

This is probably an edge case for "regular" .4 nozzles because most of the time this limit is not reached at humane printing speeds. I discovered this because I sliced for a 0.8mm nozzle and tried to limit the speed for a more beautiful surface finish and because I'm using Fiberlogy Fiberwood.

In my opinion the filament throughput limit should not be overridden in any way.

Maybe I'm completely wrong though and this is intended behavior for the dynamic overhangs, because it outputs less material and can thus move faster maybe?

This issue's cause could also be the cause for https://github.com/prusa3d/PrusaSlicer/issues/11983

### Project file & How to reproduce

Steps to reproduce:
- Set "Filament Settings" -> "Advanced" -> "Max volumetric speed" to something really low
- In the GCode view set the legend to "speed (mm/s)
- you see that the dynamic overhang parts are printed faster than the rest of the model, which is limited by the max filament volumetric speed

[image]

[3DBenchy_lowvol.zip](https://github.com/prusa3d/PrusaSlicer/files/14074577/3DBenchy_lowvol.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 11 x64

### Printer model

Prusa i3 Mk3S MMU2S

total1ty, 2024-01-28:
### Workaround
disable dynamic overhang speeds

kubispe1, 2024-01-30:
Thank you for investigating this issue and providing us with a detailed description. We will work on fixing this issue.
SPE-2128

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena