prusa3d/PrusaSlicer #12149: [multi material] Impossible to adjust material seam overlap

viesturz, 2024-01-28:
### Description of the bug

The two material seams are generally a weak adhesion point. For multiple reasons, like under extrusion, not perfect nozzle alignment, material differences.

Today I'm printing with 2 TPU filaments, resulting in a seam that can be easily torn apart.
I tried changing the model to have 0.2 mm overlap, but prusaslicer just reverts that.
I looked for any settings that allow specifying an overlap, or at least stop removing the overlap in the model with no results.

### Project file & How to reproduce

[BadgeMulticolour.zip](https://github.com/prusa3d/PrusaSlicer/files/14076036/BadgeMulticolour.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Win11

### Printer model

voron 2.4

clfaye, 2024-01-29:
+1  
I am printing a model with a perimeter of polycarbonate glass fiber sandwiched between two perimeters of clear polycarbonate.  The perimeters can easily be torn apart like the OP stated.  Ironically, layer adhesion, which is usually the weak point, is great between these materials.  So, they are compatible but just not quite overlapping enough to bond well between perimeters.  I am experimenting to see if a high extrusion multiplier will help - but that makes the print messy in other areas.  Another way to handle this would be to have extrusion multiplier as an object setting.  You could split models up and only use the multiplier where necessary.  But, that probably wouldn't work for everything.

clfaye, 2024-01-31:
I can confirm that there is a bug where Prusa Slicer is not calculating perimeters correctly for multi part models which is resulting in weak adhesion between model parts.  Here are shots of a model sliced as a single part and as a multipart to compare how Prusa Slicer is calculating line widths.  In the single model version, Prusa Slicer creates three perimeters.  The exterior perimeters are about 1.2mm and the interior perimeter is about 0,9mm with a little extra at the corners.
[image]

When the model is divided into three separate parts, Prusa Slicer calculates much lower line widths.  The two exterior parts are given about 0.9mm line widths and the interior part is given the same, about 0.9.  This results in much less material being extruded and little or no adhesion.
[image]

[project files.zip](https://github.com/prusa3d/PrusaSlicer/files/14115606/project.files.zip)

closed 2026-09-29 by prusaslicer-bot[bot]