prusa3d/PrusaSlicer #12157: Prusa Slicer Error Loading obj file.

briansturgill, 2024-01-30:
### Description of the bug

No matter what obj file I load (all generated by trimesh), I get prusa slicer reporting 3 open edges.
Even this simple .obj file (of a cube) gets this error.
I can't see anything wrong with this file. It has the expected 8 vertices and 12 faces.
Meshlab doesn't see a problem with it.
```
# https://github.com/mikedh/trimesh
v 0.00000000 0.00000000 0.00000000 0.94117647 0.90196078 0.54901961
v 0.00000000 0.00000000 10.00000000 0.94117647 0.90196078 0.54901961
v 0.00000000 10.00000000 0.00000000 0.94117647 0.90196078 0.54901961
v 0.00000000 10.00000000 10.00000000 0.94117647 0.90196078 0.54901961
v 10.00000000 0.00000000 0.00000000 0.94117647 0.90196078 0.54901961
v 10.00000000 0.00000000 10.00000000 0.94117647 0.90196078 0.54901961
v 10.00000000 10.00000000 0.00000000 0.94117647 0.90196078 0.54901961
v 10.00000000 10.00000000 10.00000000 0.94117647 0.90196078 0.54901961
f 2 1 5
f 3 5 1
f 2 4 1
f 4 2 6
f 4 3 1
f 4 8 3
f 6 5 7
f 6 2 5
f 7 5 3
f 8 7 3
f 8 4 6
f 8 6 7
```

### Project file & How to reproduce

Just load the obj file onto the platter...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+linux-x64-GTK3

### Operating system

Ubuntu 22.04

### Printer model

MK4 and MK3S+

Jony01, 2024-01-30:
Thanks for reporting an issue.
Application need line end after last face(triangle indices) at the moment.
For fast fix, add new line(enter) after last face.
[not_working_cube.obj](https://github.com/prusa3d/PrusaSlicer/files/14095235/not_working_cube.obj.txt)
[working_cube.obj](https://github.com/prusa3d/PrusaSlicer/files/14095241/working_cube.obj.txt)

briansturgill, 2024-01-30:
Thanks for the workaround!

lukasmatena, 2024-09-12:
Fixed in [2.8.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases). Closing.

closed 2024-09-12 by lukasmatena