prusa3d/PrusaSlicer #12163: Layer height rounding error in top layer?

ps-igel, 2024-01-30:
### Description of the bug

Slicing an stl file creates a weird structure on the top layer of the object that actually is just flat.

It is supposed to look like that:
[image]

But it looks like that:
[image]

Happens using Default PrusaSlicer 0.2 Quality profile. Changing the layer height as little as 0.201 or picking a different layer height profile resolved the issue.

### Project file & How to reproduce

Load stl file into slicer, use 0.2 layer height -> slice.
3mf and stl file are attached.
[strap.zip](https://github.com/prusa3d/PrusaSlicer/files/14103942/strap.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1+win64

### Operating system

Windows 10 64-Bit

### Printer model

Prusa Mini

mix579, 2024-01-30:
My first suspicion was that the top surface is not flat, and that was confirmed when I loaded the STL into Fusion. Top and bottom surface are not parallel, which leads to the top surface being slanted. Based on your layer height, PS may or may not try to put another layer in to compensate.
See picture: https://i.imgur.com/AQHmwzp.png

ps-igel, 2024-01-31:
> Top and bottom surface are not parallel

Which is weird, because both surfaces are constructed as parallel in fusion. I attached the original fusion file.

What I noticed in PrusaSlicer is that it makes a different if I put the model on its flat side by using the "place on face" function of if I rotate it by hand. Rotating by hand works and the top surface is sliced ok.

The place on face does not seem to rotate it 100% correctly to the plane. If I use the cut function and set the cut position to 0 it will still mark a tiny bit to cut off from the model.

[image]

neophyl, 2024-01-31:
I had a look at the vertices level at the stl in blender.  Some of the verts around the holes are 0.00000003 off zero on that bottom.  You also have some thin faces and a weird sticking out vert around the back section but that doesnt effect the placement.  I think its just enough to throw the place to face tool off.  
If I flatten the verts in Blender and export then when loaded in it places to the face properly and slices like so -
[image]

[Strap_blender_export.zip](https://github.com/prusa3d/PrusaSlicer/files/14115207/Strap_blender_export.zip)

mix579, 2024-01-31:
> I attached the original fusion file.

I don't see it. The zip file in the original post contains an STL and a 3MF file but no .f3d file, which is the only format that let's us clearly see where things went awry as it contains the steps in the Fusion design.

mix579, 2024-01-31:
Mmh, I don't see anything wrong in the f3d file. However, before extrusion 7, if I tesselate the body and then convert the mesh back to brep, the two surfaces of the strap I see an angle of 0, if you do it after extrusion 7, I get a slight angle between them. Seems like Fusion is doing something odd here. The two lines defining the surfaces in sketch 2 are actually not constrained to be parallel, but even if I apply the parallel constraint, it's not changing anything.

ps-igel, 2024-02-01:
Ok so, I think it is safe to say that this is not a slicing issue. I'll tinker around in Fusion to see whats going on.
Thanks all for investigating. 
Closing this issue now.

closed 2024-02-01 by ps-igel