prusa3d/PrusaSlicer #12169: Build fails on MacOS: 'Target "wxcore" links to: CMath::CMath but the target was not found'

sethml, 2024-02-01:
### Description of the bug

Attempting to build on MacOS following the [How%20to%20build%20-%20Mac%20OS](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md) instructions. Building deps fails:

```
% git clone git@github.com:sethml/PrusaSlicer.git
% cd PrusaSlicer
% git log
commit 97c3679a37e9ede812432e25a096e4906110d441 (HEAD -> master, origin/master, origin/HEAD)
Author: rtyr <36745189+rtyr@users.noreply.github.com>
Date:   Mon Jan 8 11:09:16 2024 +0100
...
% cd deps
% mkdir build
% cd build
% cmake ..
...
% make
...
-- Which libraries should wxWidgets use?
    wxUSE_STL:        OFF  (use C++ STL classes)
    wxUSE_REGEX:      OFF  (enable support for wxRegEx class)
    wxUSE_ZLIB:       sys  (use zlib for LZW compression)
    wxUSE_EXPAT:      sys  (use expat for XML parsing)
    wxUSE_LIBJPEG:    sys  (use libjpeg (JPEG file format))
    wxUSE_LIBPNG:     sys  (use libpng (PNG image format))
    wxUSE_LIBTIFF:    sys  (use libtiff (TIFF file format))
    wxUSE_NANOSVG:    sys  (use NanoSVG for rasterizing SVG)
    wxUSE_LIBLZMA:    OFF  (use liblzma for LZMA compression)
    wxUSE_LIBSDL:     OFF  (use SDL for audio on Unix)
    wxUSE_LIBMSPACK:  OFF  (use libmspack (CHM help files loading))

-- Configured wxWidgets 3.2.0 for Darwin-23.3.0
    Min OS Version required at runtime:                macOS
    Which GUI toolkit should wxWidgets use?            osx_cocoa
    Should wxWidgets be compiled into single library?  OFF
    Should wxWidgets be linked as a shared library?    OFF
    Should wxWidgets support Unicode?                  ON
    What wxWidgets compatibility level should be used? 3.0
-- Configuring done (0.2s)
CMake Error at build/cmake/functions.cmake:454 (target_link_libraries):
  Target "wxcore" links to:

    CMath::CMath

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  build/cmake/lib/core/CMakeLists.txt:61 (wx_lib_link_libraries)

-- Generating done (0.2s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
make[2]: *** [dep_wxWidgets-prefix/src/dep_wxWidgets-stamp/dep_wxWidgets-configure] Error 1
make[1]: *** [CMakeFiles/dep_wxWidgets.dir/all] Error 2
make: *** [all] Error 2
```

I've tried to figure out what's going on, but my cmake-fu is too weak. Help?

### Project file & How to reproduce

MacOS 14.3 (23D56)
Apple M2 Pro
cmake version 3.28.2
XCode Version 15.2 (15C500b)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

97c3679a3

### Operating system

MacOS 14.3 (23D56)

### Printer model

irrelevant

barzto, 2024-02-05:
I had same issue with recent cmake versions 3.28.0--3.28.2, but was able to build deps and Slicer with CMake 3.27.9. 
Unfortunately there's no simple way how to install older brew package (only some packages has multiple versions in brew, which is no the case for cmake), so I'd suggest to download and install cmake 3.27.9 [directly from Kitware](https://cmake.org/download) or  [here is the direct link to the universal DMG](https://github.com/Kitware/CMake/releases/download/v3.27.9/cmake-3.27.9-macos-universal.dmg).

Once you have copied the CMake app into the `/Applications` you may invoke it like this:

```
cd build
/Applications/CMake.app/Contents/bin/cmake ..
make -j6
```

If you have dropped the CMake app to custom location, make sure there's no space in the path.

jmerdich, 2024-04-07:
Note this is a [libTIFF bug](https://gitlab.com/libtiff/libtiff/-/issues/625)... the new version of cmake just happens to be the first to try and use libTIFF's new packaging code that's present in the static-bundled version in the deps. It also applies to more than Macs-- I hit it on Ubuntu with a recent enough cmake.

There are a couple of workarounds:

 - Downgrade to cmake 3.27.x.
 - Modify `wxWidgets.cmake` to unset wxUSE_LIBTIFF (or after https://github.com/prusa3d/PrusaSlicer/pull/12559 is merged, build deps with `-DEP_WX_ENABLE_TIFF=OFF`). The slicer build will not be able to work with tiff files.

MaleNurse, 2024-04-08:
@jmerdich I downgraded all the way to 3.27.0 and am still getting the same errors. Which version did you use to successfully build yours with?

barzto, 2024-04-08:
@MaleNurse I managed to build Slicer with CMake 3.27.9 (should  be the last 3.27.x version). Did you removed original deps/build directory before building with CMake 3.27.x (to be sure there is no invalid cached value stored)? Are you using the 3.27.x version for both building deps and the app itself?

MaleNurse, 2024-04-08:
@barzto Unfortunately, yeah. Tried w/ 3.27.9, reverted further back to 3.27.0, trashing the build folders each time. Then I completely wiped and re-cloned the project, but nothing seems to be working. Only thing I haven't tried is modding wxWidgets.cmake since I'm just trying to retain all the intended functionality. The commands I'm using are:

`/Applications/CMake.app/Contents/bin/cmake ..` for the deps and

`/Applications/CMake.app/Contents/bin/cmake .. -DCMAKE_PREFIX_PATH="$PWD/../deps/build/destdir/usr/local"` for the app

`/Applications/CMake.app/Contents/bin/cmake --version` gives me `cmake version 3.27.0` right now

inistor, 2025-12-18:
Fixed upstream; see: https://gitlab.com/libtiff/libtiff/-/merge_requests/764

closed 2026-09-29 by prusaslicer-bot[bot]