prusa3d/PrusaSlicer #12174: Overlap of Supports and Brims in PrusaSlicer

Shah-Priyanshu, 2024-02-01:
### Description of the bug

When arranging multiple objects on the print bed in PrusaSlicer, I have encountered an issue where the automatically generated supports and brims are overlapping each other. This results in the prints being joined together by the brims and/or supports, which is not the intended outcome. The issue presents itself when the objects are positioned close to each other but are not merged. However, when I merge the objects, the slicer correctly separates the supports and brims for each object, indicating that the overlap issue is bypassed.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Project file & How to reproduce

Steps to Reproduce:

1. Arrange multiple objects on the print bed in PrusaSlicer close enough that their supports and brims would naturally overlap.
2. Enable supports and brims in the print settings.
3. Slice the model and observe the resulting preview.

Expected Behavior:
Supports and brims should be generated individually for each object without overlapping, regardless of whether the objects are merged or not.

Actual Behavior:
Supports and brims overlap when objects are positioned close to each other but not merged. This overlap is eliminated when objects are merged.

Workaround:
Currently, merging the objects seems to prevent the overlap of supports and brims. However, this is not always a viable solution, as merging might not be suitable for all printing scenarios.

[Files.zip](https://github.com/prusa3d/PrusaSlicer/files/14131111/Files.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 11

### Printer model

Original Prusa i3 MK3S & MK3S+

neophyl, 2024-02-02:
Please search open and closed issues before creating new ones.  This has been reported many many times.  It is expected operation.  If you don't want them sliced individually you need to merge them into a single object.  
The currently open issue where these are collected is https://github.com/prusa3d/PrusaSlicer/issues/316 

Please read that issue and some of the linked issues if you want to understand more about why this happens.

lukasmatena, 2024-02-06:
Closing as a duplicate of #316. PrusaSlicer slices all the objects separately (which means that it can speed up reslicing when only one of the objects change), meaning that you can get collisions like this. Recent PrusaSlicer versions emit a warning to alert the user when this happens.

Snuff1eupagus, 2024-08-31:
@Shah-Priyanshu 
The example you provided is seemly, to me anyway convoluted, however these tutorials may help with your needs.

Selective Blocking
https://www.youtube.com/watch?v=6CRqNk_JUj4

Selective Adding
https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s

Regards