prusa3d/PrusaSlicer #12182: Feature Request - Match System Theme (light mode/dark mode)

phazejeff, 2024-02-03:
Dark Mode tab in the preferences should have a "Match System Theme" option which uses light mode or dark mode based on if the system is in light mode or dark mode. I personally have my system automatically switch between light mode and dark mode based on if it's day or night, and most of my programs follow the system theme. It'd be nice for this one to do it as well.

neophyl, 2024-02-03:
You mean like this -
[image]
Thats the menu on a windows 10 install.

Might help if you actually included some details about your system, like what OS it is, what PS version you are using etc.  The basics.

phazejeff, 2024-02-16:
> You mean like this - [...]

No, that doesn't do what I was explaining. I am also on WIndows 10. I didn't feel like trying to explain it further or waiting around so I just made the change myself at pull request #12267

closed 2026-09-29 by prusaslicer-bot[bot]