prusa3d/PrusaSlicer #12183: Gaps in the infill when using Arachne

AlexDaniel, 2024-02-03:
### Description of the bug

When using Arachne, it ends up creating gaps in the print:

[image]
[image]

### Project file & How to reproduce

* STL file: [PlayerTokens_x2.STL.zip](https://github.com/prusa3d/PrusaSlicer/files/14149110/PlayerTokens_x2.STL.zip)
* The original model is under CC BY-NC and can be found here: https://cults3d.com/en/3d-model/game/alchemists-king-s-golem-expansion-boardgame-insert

Printing/Slicing instructions:
* Slice with the default settings. I used 0.10mm layer height.
* Arachne in the preview seems to show thicker/weirder lines for the infill, and it looks like it is supposed to fill everything correctly, so the problem is not really visible in the preview, but is blatantly obvious in the final print.
* If you switch to Classic (Print Settings/Layours and perimeters/Perimeter generator → Classic), then all of the problematic places are covered with “Gap fill” after slicing.

Arachne:
[image]
Classic:
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Debian

### Printer model

Prusa i3 MK3S+

AlexDaniel, 2024-02-03:
To be clear, these are not the only two places that have gaps. Basically, any place where you get “Gap fill” generated in the classic mode, will actually end up being a true gap between the perimeters if you use Arachne.

kylek29, 2024-02-04:
I've come across this a lot and it's why I stopped using Arachne. From my experimentation, it happens when you have a fillet, so you have an arc meeting with a straight. 

I tried just about every combination of gap setting adjustment I could think of, there's just none that could get it to fill those spots. The fact it's visible in the simulation (sliced view) tends to show that A) that view pretty dang spot on (well done lads) and B) it's not so random unaccounted for GCode issue (like filament shrinkage).

AlexDaniel, 2024-02-05:
@ kylek29 It's intentional. Arachne uses varying extrusion widths to fill the gaps, so the “gap fill” pass is supposedly not required. In practice, however, there's still a gap, so there's a mismatch between the expectations of Arachne (i.e. fill width) and the real world.

And yes, I stopped using Arachne for that reason too. It's a shame because the classic mode has its own issues which are often even more ugly, but at least I'm not getting these through holes because of gaps.

Tupson444, 2024-02-09:
Increasing "Perimeter transitioning threshold angle" might solve this problem (Print Settings > Advanced, under Arachne perimeter generator). See my comment on this issue: https://github.com/prusa3d/PrusaSlicer/issues/8979#issuecomment-1272273652

AlexDaniel, 2024-02-10:
@Tupson444 the preview has no gaps though, so I don't see how changing any setting can affect it.

Tupson444, 2024-02-10:
There is a difference in the preview. Here is the default settings of 10° (as far as I can see, this is the gap that would show in the printed model):
[image]

and here is when "Perimeter transitioning threshold angle" is changed to 35°:
[image]

AlexDaniel, 2026-09-02:
@kylek29 do you have a quick way to test it now? I feel like I haven't come across this issue in the last year or so, but I'm not sure.