prusa3d/PrusaSlicer #12184: Huge blob in wipe tower

ruvilonix, 2024-02-03:
### Description of the bug

I'm doing multiple colors in a single layer with one extruder. I enabled the wipe tower, but for some reason it creates a massive blob when it begins the wipe tower actions, which gets in the way of the hotend. 

Here's the relevant G-Code after the tool change:

```
M600; Filament change
T1
; Filament gcode

G1 F9000
G1 X183.175 Y212.143
G1 Z.2
G4 S0
; CP TOOLCHANGE LOAD
G1 E18.0000 F180
G1 X213.968 E63.0000 F821
G1 X183.175 E9.0000 F575
G1 Y212.643
; CP TOOLCHANGE WIPE
G1 X174.843 E0.3737 F396
M220 R
G1 F9000
G4 S0
G92 E0
; CP TOOLCHANGE END
;------------------
```

### Project file & How to reproduce

[gear.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14149522/gear.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.1+linux-x64-GTK3

### Operating system

Ubuntu 23.10

### Printer model

Ender 3 S1

rtyr, 2024-02-05:
If you are using "M600 hack" than you should disable "single nozzle multi material". Single nozzle material parameter expects actual multi-material capable printer (multi-material unit, Y-splitter or something like that).

closed 2024-02-05 by rtyr

ruvilonix, 2024-02-06:
Nice, thanks. Two different tutorials on the M600 trick said to use that setting, and I couldn't find much documentation about it. I haven't tried printing yet with it off, but the slicer's g-code looks much more normal.