prusa3d/PrusaSlicer #12188: Avoid organic supports during travel

fjddfopji, 2024-02-04:
**Is your feature request related to a problem? Please describe.**
I already have avoid crossing perimeters on. This setting only applies to the printed part not the organic support towers. The gcode has the printer print some of the organic supports, then it crosses over a support it already printed for that layer. This means the nozzle contacts the support and knocks it off, causing the print to fail. 

**Describe the solution you'd like**
There should be an option to avoid crossing already printed supports.

**Describe how it would work**
The printer should avoid supports where it has already printed on that layer.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.
[image]
The issue can be seen in the upper left hand corner

Frankenmint, 2024-02-04:
Im facing this issue now. Even if I set travel lift to .6mm it still hits the supports: https://www.youtube.com/shorts/rymvpPkQneg

Joe-0911, 2024-02-19:
I'm going to add my voice to the issue. I've all but given up on organic supports for this reason.  Z-hop, for some reason, also doesn't seem to resolve the issue.

Example of a particular problem travel path can be found in the image. 

[image]

evilhamsterman, 2024-02-29:
This was mentioned before in #10077 and declared fixed but I agree it probably isn't

Tupson444, 2026-09-18:
maybe still relevant