prusa3d/PrusaSlicer #12190: Embossed text is not updated when changed directly in the 3mf file

MnlSx, 2024-02-04:
### Description of the bug

When text is embossed into an object with the "Emboss text" tool, the string that represents the value of the text is safed in the "Slic3r_PE_model.config" file thats part of the .3mf archive when the project is safed. 
This parameter can conviniently be changed outside prusaslicer in any texteditor or by custom scripts. This would be a great way to print a lot of objects with different texts embossed to them (for example lables or nametags) without having to change the text by hand in Prusaslicer which is very inconvinient.
Unfortunately, Prusaslicer does not automatically updated the text that is embossed into the objext when the edited project is opened. The edited text is visible in the edit window of the emboss text tool, but is only updated to the object when changes are made to the text in this windows by hand. 

### Project file & How to reproduce

### Steps to reproduce:
- Open Prusalicer 2.7.1
- Import an object eg. a box
- Use the "Emboss Text" tool to add some text to the object
 
[image]
- Safe the project into a .3mf file
- Unzip the .3mf file
- Open the "Slic3r_PE_model.config" located inside the "Metadata" folder with a texteditor
- Locate the parameter for the text you have embossed before, it should look like this:  (text="OLD")
[image]
- Change the value to some different string
[image]
- Safe the file
- Zip all the folders extracted from the .3mf file before and change the file extrension to .3mf
- Open the file in prusaslicer
- **UNEXPECTED BEHAVIOR** The text embossed to the object has not changed. However, the text in the edit embossed text window has changed.
[image]
- When additional changes to the text are made by hand in the edit embossed text window, the embossed text changes as expected
[image]

**EXPECTED BEHAVIOR**
When a project file with embossed text is opened, The embossed text should update if the string representing this text in the Slic3r_PE_model.config has changed

[Examples_Embossed_Text_Change.zip](https://github.com/prusa3d/PrusaSlicer/files/14157307/Examples_Embossed_Text_Change.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 10

### Printer model

Prusa XL 5T, Prusa MK3S+

Jony01, 2024-04-22:
The data connected to text in `Slic3r_PE_model.config` are for reconstruct stored 3d shape(e.g. in case of surface move).
You missinterpreted meaning of stored metadata in 3mf.

In my opinion: It is unwanted behavior to change 3d model imidiatelly after opening 3mf.
(In general it will make more trouble than benefit)

Please, describe more your use case why one need it.

MnlSx, 2024-04-23:
> Please, describe more your use case why one need it.
I think I understand the meaning of the stored metadata. My intention was to to reuse this already avalible feature for another purpose.

My use case is that I would like to print hundreds of nametags, all with different names on them. Changing the name in the .3mf metadata could easily be automated, while changing the text in prusaslicer would need to be done by hand. 
But such a function would also be usefull to emboss serial numbers or batch numbers in parts - something I had wished for in the past.  

I get that automatically changing the 3d model after opening the file could have unwanted side effects. Maybe it would be an option to have a function in the edit menu that allows the user to update the model in regard to the metadata? I see no harm in that and it might enable advanced users to do other "tricks" by changing the 3mf file. 

Of course, the cleanest solution would be to have a function in prusaslicer that accepts a list of strings and automatically embosses them to the objects. But this feels like a pretty complex new feature, that might need to much rescources to be implemented.

Jony01, 2024-04-24:
From my point of view: 
For this purpose will be better to use command line interface and add option to change text.

But this solution brings several issues, which must be solved first:

- [ ] How to load font, now is using wxWidgets (frontend)
- [ ] How to select the text in model (some identifier of text in 3mf)

MnlSx, 2024-04-26:
I would be open to use the command line interface for this usecase. 
Do you see any chance that this feature will be implemented in the near future, either in the command line interface or the gui? I would really like to use prusaslicer for this purpose, as doing it in cad brings a lot of other problems I would like to avoid.

choller, 2025-10-06:
@Jony01  I'd be happy to implement your console approach instead. However, it was my understanding, that the specified emboss feature is not available through console only, or am I missing something? I can certainly try to implement it the way you describe and in fact, I would find that much cleaner as well.

throwaway8787, 2026-09-02:
@prusaslicer-bot - you suck.