prusa3d/PrusaSlicer #12193: Re-Naming objects after slicing issue.

Sk4nk-Hunt, 2024-02-05:
### Description of the bug

If you slice a objet(s). Then re-name it. Then export it. The new object names WILL NOT export. 
You NEED to Re-Slice, for it to apply when exporting.

Expecting for slicer to inform me to "Re Slice"  the scene. 

This is important for when trying to cancel an object mid print on the XL, and the naming being all wrong on the objects. 

### Project file & How to reproduce

1. Slice an objet(s). 
2. Then re-name objects. 
 2.1.(DO NOT RE-SLICE HERE AGAIN)
3. Then export it (bgcode I did)

4. Testing bgcode = open print file on XL Printer -> cancel objects section on LCD screen when printing. 

The new object names WILL NOT export when viewed on 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Mac

### Printer model

Prusa XL 5T

lukasmatena, 2024-02-06:
Thanks. This was already mentioned in #8059, the fix is on the way (SPE-2139).

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena