prusa3d/PrusaSlicer #12202: Feature to a bgcode

marcos-ramalho, 2024-02-07:
The bgcode file does not allow reading the adjusted parameters.

**Describe the solution you'd like**
The coordinate data and commands do not need to be read by users, but the parameters set for each bgcode should appear in the file without being compiled.

**Describe how it would work**
The gcode shows the adjusted parameters, which allows you to use it to compare with other files or even with PrusaSlicer values, whereas bgcode does not show it.
  Sometimes, in the testing phase to get the best adjustment or because we haven't saved a good adjustment, in gcode we can open the file and see what was adjusted, whereas in bgcode this is not possible.
I don't see any problem with bgcode being compiled, but the parameters could be readable.

**Describe alternatives you've considered**
The bgcode keeps the compiled code that we cannot read, but at the end of this code the parameter settings are displayed in a way that the user can read.

**Additional context**
Add any other context or paint drawing about the feature request here.

mix579, 2024-02-07:
Prusaslicer already allows you to convert bgcode to gcode, which will show you all the parameters, and that's only one extra step. So adding this stuff to the bgcode file would just add bloat to that file.

marcos-ramalho, 2024-02-07:
Hello MIX579, thank you for your response. I didn't know about this utility
of being able to transform bgcode into gcode.
You mean that I can take a bgcode file and convert it, but I researched a
little and didn't find how this can be done.
Can you explain to me how?

a hug
Marcos Oliveira

mix579, 2024-02-07:
Check out Prusaslicer>File>Convert>Convert binary gcode to ASCII

marcos-ramalho, 2024-02-07:
Tks
Marcos Oliveira

kubispe1, 2024-02-08:
Also you can read g-code lines of bgcode in G-Code Viewer. 
I am closing the issues. 
Thanks

closed 2024-02-08 by kubispe1