prusa3d/PrusaSlicer #12210: UX redesign.

GAZ082, 2024-02-09:
Hi. Would be nice to have an interface redesign where you do not need to leave the main screen to tune the most used settings.

OrcaSlicer/BambuSlicer do this with a tabbed column where you have logically grouped parameters.

If the dev team does not want to mimic other slicers, ok, but a GUI revamp would be nice, not just a touch here and there.

neophyl, 2024-02-09:
Try changing your preferences to select Non Modal window.  (Which has been available for quite some time).

[image]
That way ALL the settings can be seen in a separate window and accessed without leaving the plater view.

That allows you to use it like so-
[image]

GAZ082, 2024-02-10:
Thanks, but in a 1920x1080 screen you cannot fit both, even after shrinking  both windows fully.

At least in Gnome.

closed 2026-09-29 by prusaslicer-bot[bot]