prusa3d/PrusaSlicer #12219: Missing retracts on a layer change when a toollhead moves from one object to another

rednaxal, 2024-02-10:
### Description of the bug

The latest 2.7.2-alpha2 is having a serious issue - it is missing retraction on a layer change in case of multiple objects on a table, i.e. when a layer starts not at the same place where it ends like on a simple objects with a single seam.
As a result many of my prints stared to have a loooooot of stringing when sliced in 2.7.2-alpha2, so had to rollback to 2.7.1. 
[image]
Also for some reason it moves toolhead to a certain Z coordinate twice. 

### Project file & How to reproduce

[bug.zip](https://github.com/prusa3d/PrusaSlicer/files/14230799/bug.zip)
Just put two objects on a table and slice it then look into gcode on a layer change. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2-alpha2

### Operating system

Windows 10

### Printer model

KP3S Pro

freakydude, 2024-02-11:
Same here. Slicing a simple retraction test model (2 cylinders,3mm, 25mm height) 
Instead of 2 Retractions and 2 Deretractions there only 1 of each on all layers.

Model 
[image]

Prusa 2.7.1 Release
[image]

Prusa 2.7.2-Alpha2 
[image]

[2.7.2-alpha2-retraction-bug.zip](https://github.com/prusa3d/PrusaSlicer/files/14233137/2.7.2-alpha2-retraction-bug.zip)

With best regards
freakyDude

virtualcomputerguy, 2024-02-11:
Same issue. Seeing it also not retracting on single large parts as well. Tons of stringing and visible artifacts on parts. Same issue on multiple printers. I do see the retraction line in the g-code, but it is not performing a retraction.
[image]

rednaxal, 2024-02-12:
@SachCZ 
So you mean we now have to keep the "Retract on layer change" always enabled for the retraction to work properly? 
But this setting doesn't make any sense for a simple prints where a layer starts at the same place where it ends (Imagine a simple 2 walls box for example). So retractions there aren't necessary at all but with that setting enabled it will be happening and possibly affecting the look of a seam and the print time. 
Also why do we need a setting which when disabled in most cases results in a bad prints? :) And for some reason it is located not in a print settings but in a printer settings, i.e. shouldn't be changed depending on a model to print. It makes no sense to me, sorry. 

In my opinion this setting shouldn't affect the retraction for horizontal travel movements at all. And should apply only to a layer change (which from a user's perspective is supposed to be only a Z change, like in 2.7.1). While every time there is a horizontal travel of a toolhead (not less than retract_before_travel mm) there MUST be a retraction regardless of whether it is a part of a layer change code or a whatever else.

SachCZ, 2024-02-12:
I see your point about horizontal travels, we will look into it - **possibly returning to the previous behaviour in next public testing release**. Thank you for your feedback, it is much appreciated. SPE-2148

To continue using 2.7.2 alpha2, as a workaround, you can keep "Retract on layer change" on for most of your prints and turn it off for special prints like boxes with 2 walls. I agree that in this case having to change printer settings based on model you are using is sub-optimal, please bear with us while we iron out some of the quirks of the next PrusaSlic3r release.

freakydude, 2024-02-12:
@SachCZ 
Indeed, the headline of this issue points to two issues:
- Retract on LayerChange
- Retract/Deretract before and after travel while remaining on the same layer.

The functionality of both seems to be changed. 

I was mainly talking about the retraction/deretraction on the same layer and like the opinion of @rednaxal I would also expect, that the option "retract on layer change" does not influence the retract on the same layer. As as can be seen in real prints it's very effective to remove/minimize stringing. I would be surprised if you were to drop this.

best regards

SachCZ, 2024-02-12:
I cannot say that I completely understand you @freakydude. I have not been able to reproduce the issue on same layer. Maybe we are not on the same page here when it comes to the terms being used.

The gcode is generated in the following way:
- layer 1 gcode
- layer change gcode
- layer 2 gcode
In alpha 2, layer change starts at last point of layer 1 and ends at first extrusion point of layer 2, meaning there is a travel. This travel does have or does not have retraction based on the setting "Retract on layer change". **I am pretty sure that "Retract on layer change" does not influence any retraction/deretraction on the same layer.** Maybe you mean that you would expect a retraction for any travel, even if it is part of layer change (as already stated by @rednaxal).

SachCZ, 2024-02-12:
@virtualcomputerguy thank you for your time. My guess is that you also have "Retract on layer change" turned off. Is that the case? Please do your testing with this setting turned on for 2.7.2 alpha 2.

virtualcomputerguy, 2024-02-12:
OK. I can try that. I have never needed that for long travel moves during the same layer. I do use that setting, but not on this model. I will enable it and try again. But I would hope that this does not go into production like that. You should not need to enable retract on layer change to have retraction working on the entire layer.

I am seeing tons of stringing on long travel moves and none at the end of the layer. It sounds like that setting will take care of one of the issues. Will update after testing again.

freakydude, 2024-02-12:
@SachCZ 

Ok I try to clarify - maybe we talking about the same - or not. I think after that its clearer.

I did another stringing test with 4 3mm towers instead of 2 to see the behavior more clearly. I filtered just 2 slices at 15/15.2mm z - but its the same for all - of course. The new model is attached here if needed: [prusa-2.7.2-alpha2-retract-travel.zip](https://github.com/prusa3d/PrusaSlicer/files/14250460/prusa-2.7.2-alpha2-retract-travel.zip)

Prusa 2.7.1 with Retraction on Layer Change - OFF -
The only think is missing, is the retraction after the layer changed ;-) but thats the inactive option

[image]

Prusa 2.7.1 with Retraction on Layer Change - ON -
I don't miss anything, here are all retraction on layer change and before and after travel moves
[image]

Prusa 2.7.2-alpha2 with Retraction on Layer Change - OFF -
In this example, I miss also the deretraction after the first travel move. That's the additional issue (or the issue you already talking about ;-) 
[image]

Prusa 2.7.2-alpha2 with Retraction on Layer Change - ON -
I don't miss anything, here are all retraction on layer change and before and after travel moves

[image]

And, whats interesting too, the move direction changed to counter clockwise (same profile as in 2.7.1)

Hop that helps.

best regards

freakydude, 2024-02-12:
I agree, this makes sense @rednaxal. In my mind, a layer change was that the nozzle moves to pos xy, printing/extruding. Then z is increased and printing continues at pos xy,z+1 with printing. So why should I retract? So I disabled it like you ;-)

but like with the alpha, if you increase z first and then (move) to a pos x2,x2 - then a retraction makes more sense.

Personally I would do it the first way (because I don't know the advantages of the second way).

SachCZ, 2024-02-14:
Hi, you are both right. Thank you @freakydude for the detailed analysis, I think it is in line with the conclusions already stated. We are working on changing it, as the new behaviour clearly causes confusion and in combination with  "Retract on layer change" being on the printer tab it really does not make sense.

Just to make a slight correction to your reasoning about layer changes @freakydude. It has always been:
1. nozzle finishes printing layer at x,y,z
2. layer change happens: nozzle moves up x,y,z+1
3. nozzle moves to the first position of the next layer x2,y2,z+1
The difference is that in 2.7.2 alpha 2 steps 2. and 3. are both part of the layer change and are thus "Retract on layer change" affects them as a whole. For context, this has been introduced with the new ramping layer change feature, where this mental model of layer change being single move up does not make sense anymore. Anyway we are working on preserving the previous behaviour.

freakydude, 2024-02-14:
@SachCZ Thank you for the detailed explanation and good luck separating the concerns. 

By the way, you are welcome to contact me if you would like to discuss the mixing of bridge and overhang settings - and the associated inadequacies in bridge/overhang (fan) speed detection.

I had already opened an issue, but it's probably a bit much at once - and possibly also a mistake in my thinking ;-)

best regards

lucasavendano, 2024-02-20:
Hi, i am seeing the need to show you something, maybe its related to this, when 2.7.0 beta 1 was released i started to have a lot of stringing in all my machines, and i was unable to figure it out, almost lose my mind with hardware and filaments, retraction test, etc. In the same time i was trying orca slicer, they are doing a great job (for me), specially with things related to klipper. When they released 1.9 the problems with stringing started too. I don´t know exactly how the things works in this of "based on prusa slicer..." but i think they took some problem in PS new versions...

Recently they worked on this:

[https://github.com/SoftFever/OrcaSlicer/pull/3888#issue-2105551089](url)

I think this could be helpfull

🙋‍♂️

SachCZ, 2024-02-20:
Hi @lucasavendano, thank you for your interesting observation. Sadly, we are unable to really address this as your description of the problem is too broad. Also, please open a new issue, as this one cannot be related to your observations - it addresses an issues in 2.7.2 alpha and you problems are related to 2.7.0 beta 1. When opening a new issue, please be so kind and provide a 3mf file for which your stringing problem was not present pre 2.7.0 beta 1 and is present now after 2.7.0 beta1. Anyway, we appreciate your effort!

closed 2024-02-20 by lukasmatena