prusa3d/PrusaSlicer #12226: Issues with PrusaXL Slicing: Poor Seam Handling and Print Quality with PETG, Gaps and Unusable Prints

LuckyResistor, 2024-02-11:
### Description of the bug

I've encountered a recurring issue when slicing models for the PrusaXL IS, specifically using a 0.6mm nozzle with PETG material. The problem manifests as pronounced gaps following the seams of the model, alongside occasional filament blobs. These defects significantly compromise the print quality, to the extent that the finished products are unusable and must be discarded.

[image]
[image]
[image]

Moreover, the issue exacerbates as the printer head drags filament across these gaps, leading to irregularities along the model's perimeter. This results in areas that appear uneven or over-extruded.

I've provided a project and gcode file that consistently reproduces this issue on the PrusaXL, based on my experiences.

## Observations from Testing

Through various tests, I've noticed a pattern related to the travel distance between seams:

- Longer travel distances between seams tend to result in larger gaps, directly impacting the seam quality. Conversely, shorter travel paths show considerably improved seam appearances. This can be easily seed in the images, where the lower part has long travels between the seams and the upper part has a shorter distance.
- This issue is particularly pronounced in models featuring thin walls with dual seams. The slicer's strategy of positioning seams at opposing ends of the model contributes to extensive travel distances, exacerbating the problem.
- When I look at the nozzle, it seems a blob of filament oozes out when it moves to the new seam position. This blob sticks then at the new location and the nozzle starts moving in the print direction, initially without any filament extracted. 

## Similar Reports

There is a similar issue, describing the same problem for Prusa MINI+. I created this new issue, as the problem persists in the latest PrusaSlicer version, and also affects Prusa XL, not only Prusa MINI.

- [#11914](https://github.com/prusa3d/PrusaSlicer/issues/11914}

## Screenshot

[image]

### Project file & How to reproduce

[gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14234081/gcode.zip)
[project.zip](https://github.com/prusa3d/PrusaSlicer/files/14234085/project.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2-alpha2

### Operating system

macOS 14.3.1 (23D60)

### Printer model

PrusaXL 5T IS

LuckyResistor, 2024-02-22:
Tested PrusaSlicer 2.7.2-beta1, the problems with the seams still persists with this version.

[image]
[image]
[image]

[print.zip](https://github.com/prusa3d/PrusaSlicer/files/14374721/print.zip)

Benigans, 2024-02-22:
Hi ! I have a similar issue (on a Mini) with seams for a while. I tuned the pressure advance value (for a 0.4 nozzle and Prusament PETG it was set to 0.4, I set it to 0.23) and my seams are better now !
Bottom = before the change / Top = after the change
[image]

Maybe you can try something like this on the XL

[image]

Dilbertus, 2024-02-22:
same problem, 2.7.2-beta, MK3S+, 3.13.2-7080, really bad with 0,6mm nozzle. both classic and arachne.

michael-kerbel, 2026-09-13:
Please don‘t close this, this is still a very serious issue for me when printing PETG on my XL. I found no sensible workaround.