prusa3d/PrusaSlicer #12227: Pause print or color change happens one layer early

yeayer, 2024-02-12:
### Description of the bug

When adding pause print or color change in sliced object, the height bar on the right is one layer off. In attached picture, you can see that the height bar position control is exactly between the color changes, however, you can see that one layer is already printed in different color. This destroyed my print. Print pause and color change should be one layer later.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.0+linux-x64-GTK3-202311231454

### Operating system

Linux Mint 20.2

### Printer model

Prusa MK4

lukasmatena, 2024-02-12:
I don't see any issue. Adding color change to layer 4.85mm has always meant that the G-code is placed at the beginning of the layer. And as far as I can tell from the screenshot, the vertical slider, legend and the preview all match.

> you can see that the height bar position control is exactly between the color changes

There is just one color change, I don't know how anything could be "between" anything.

yeayer, 2024-02-12:
Well, the slider on the right side shows that 4.55-4.85 is color1 (orange). Which is not, it is already printed in color2 (yellow). It is very confusing, as I would assume that layer starting on 4.55 will be the last one printed in first color. When looking at the slider, 4.55-4.85 is not color 1. it is already color2.

tjstansell, 2024-02-29:
I have to agree with @yeayer here.  I've only just started messing with multi-color prints and find the layer gauge on the right a bit confusing.  I think the problem is that the tick marks don't actually line up with the number printed next to them.  For instance, the first layer always shows the first layer height.  So that indicates it's the layer from 0-0.35mm, but there's no vertical color _under_ that tick mark, so it relates to the bar _above_ the tick.  But at the very top, the last tick does not have a bar above it to indicate the final color.  In my case, I'm trying to print 2 layers in white, the middle in red, and finally 2 layers in white again.  The vertical indicator shows 2 layers of white, red, then only 1 layer in white.  I think it would make more sense if the indicator always started at 0, then the tick marks show the layer height at that exact point in the color bar, just like you were reading a mercury thermometer.  The final tick mark would be the final height after the print is done, but there's nothing to select there.  As it is now, everything looks like it is off-by-1.

[image]

tjstansell, 2024-02-29:
To illustrate my point, I did this horrible mock-up of what I feel would be a more natural way of representing the gauge.  The tick marks should indicate the layer height it points to, relative to the color bar.  In this example, you know that the print is white from 0-0.35, then red from 0.35-3.95, then white from 3.95-4.35.  Perhaps the layer numbers should be presented between the tick marks instead of associated with the top height just as a visual indicator.

[image]

tjstansell, 2024-02-29:
To confuse things even more, if I switch to spiral/vase mode printing, each tick mark indicates the _bottom_ of the layer...

[image]

closed 2026-09-29 by prusaslicer-bot[bot]