prusa3d/PrusaSlicer #12230: Need an improvement on staggered seams on prusaslicer 2.7.1 or 2.7.2 alpha2

sandysasmita, 2024-02-12:
I always print an adapter for high pressure jet cleaner or a water jet pump and i always use all perimeter like 20 layers of perimeter on every object just to make it water tight. Until now i'm using prusaslicer 2.4.0 from the first date 2.4.0 were released. because in that version, the staggered seams results is the most beautiful and proper to use in my project. And after i update prusaslicer to 2.7.1 or 2.7.2 alpha2, the seams looks worse on the inner side layers but not the outside, i haven't test it on high pressure machine yet, but from what i see is the inner layer seams have a gap/hole and the water COULD BE leaked through that gap.

I need an improvement in next 2.7.2 update so the gap is closed like the outer layered of perimeter OR like the seams on prusaslicer 2.4.0

[image]
[image]

vovodroid, 2024-02-18:
Wouldn't be better use random seams for the strength? Also I have PR #11315 for concentric infill random seams.

sandysasmita, 2024-02-18:
> Wouldn't be better use random seams for the strength? Also I have PR #11315 for concentric infill random seams.

Random seams for cylindrical object makes external perimeter looks so bad, so i prefer straight line at rear or align to make it looks neat.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]