prusa3d/PrusaSlicer #12232: Thumbnail location in 3mf files

kaetir, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
The actual 3mf thumbnail is in ["Metadata/thumbnail.png"](https://github.com/prusa3d/PrusaSlicer/blob/e09f6e526a1362a95ab0945d1b28c78b77e930dd/src/libslic3r/Format/3mf.cpp#L84). This does not comply with the [Open Document specification](https://en.wikipedia.org/wiki/OpenDocument_technical_specification#Thumbnails_(directory)). Therefore, Dolphin (the KDE file explorer) does not display the Thumbnail. 

**Describe the solution you'd like**
Change the path (or add a copy of the image) at "Thumbnails/thumbnail.png". 

**Describe how it would work**
The KDE thumbnailer uses the previously mentioned path [source](https://invent.kde.org/network/kio-extras/-/blob/master/thumbnail/opendocumentcreator.cpp)

**Describe alternatives you've considered**
It is possible to use a script to change the Thumbnail placement to match the open Document specification and get the thumbnail. 

**Additional context**
N/A

3dJan, 2024-02-14:
3mf uses the Open Packaging Convention ([OPC](https://en.wikipedia.org/wiki/Open_Packaging_Conventions)), so it is not really a surprise, that it does not comply to the Open Document Specification. 

However, the thumbnail path in 3mf is not fixed. "Metadata/thumbnail.png" is just the default of lib3mf. The API of lib3mf allows to change that path.

kaetir, 2024-02-14:
My bad then, I'll go back to the KDE team to see how that can be worked out.

Thanks for your reply

kaetir, 2024-02-15:
This has been implemented recently and might not be available for your version of kio-extras.

3dJan, 2024-02-15:
You may also try https://f3d.app, it already provides an XDG compatible thumbnailer (see [Desktop integration](https://f3d.app/doc/user/DESKTOP_INTEGRATION.html)) and recently got support for 3mf, but I am not sure if kio also uses the XDG thumbnailers.

3dJan, 2024-02-15:
Or [3mfthumb](https://github.com/themanyone/3mfthumb?tab=readme-ov-file)

lukasmatena, 2024-02-15:
I believe that this is not a problem in PrusaSlicer. Closing.

closed 2024-02-15 by lukasmatena