prusa3d/PrusaSlicer #12233: When using a fuzzy shell, the slicing of the model freezes by 65% (search for support points)

ilyalivshun, 2024-02-12:
### Description of the bug

When using a fuzzy shell, the slicing of the model freezes by 65% (search for support points).
When trying to use a fuzzy shell on large 100mm by 300mm models, the slicer cuts it for a very long time. The problem only appears on larger models. At the same time, the computer processor is not loaded more than 35%, as if slicing is performed in the background. The hang-up occurs at 65%. At the same time, when slicing into an ORC, the slicer did not notice this problem.

### Project file & How to reproduce

any model

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2,7,1

### Operating system

windows 10

### Printer model

self-assembly printer

SachCZ, 2024-02-15:
Hi, thank you for reporting your issue. Would you please be so kind and provide a 3mf file where this can be clearly observed and reproduced. It would help us tremendously.

ilyalivshun, 2024-02-15:
> Привет, спасибо, что сообщили о своей проблеме. Не могли бы вы, пожалуйста, быть так любезны и предоставить файл формата 3mf, где это можно четко увидеть и воспроизвести. Это очень помогло бы нам.

Hello, yes of course. Sorry I forgot to clarify.
The thickness of the fuzzy shell is 0.05
The jitter distance is 0.1
With such settings of the outer shell, very beautiful perimeters are obtained, which perfectly hide the features of 3D printing
I understand that this deviation is too small for a fuzzy shell and will definitely require more computing power of the computer. But the hang-up only happens on the prusa slicer.
[image]
[image]
[image]

SachCZ, 2024-03-05:
Hello @ilyalivshun. Good news, thanks to the provided 3mf I was able to reproduce your issue. Even better news, in the meantime we were able to release version 2.7.2 where the issue is no **longer present!** Please download the latest version. **Thank you!**

closed 2024-03-05 by SachCZ