prusa3d/PrusaSlicer #12234: Problem with Creality Ender-3 S1 Pro config in PrusaSlicer 2.7.1

rjflory, 2024-02-12:
### Description of the bug

1st bug:
Upon initial install of the PrusaSlicer, and first run, When printer is configured for "Creality Ender-3 S1 Pro", the slicer exports BINARY .gcode, which is not supported by these printers.  By default, PrusaSlicer should assume TEXT .gcode format for Creality printers.

### Project file & How to reproduce

2nd bug:
 Once .gcode output format was manually configured as regular TEXT for my printer, I sliced a 557-layer object and saved the .gcode to SDCard.  When I printed the resulting file, the printer seemed to be told to perform continuous Auto-Calibration operations.  I aborted the calibration loop after the third iteration.  The printer indicated 6% complete when I aborted the print, before it even laid down a single millimeter of filament on the bed.  This just can't be right...

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

windows 10

### Printer model

Creality Ender-3 S1 Pro

rtyr, 2024-02-13:
> When printer is configured for "Creality Ender-3 S1 Pro", the slicer exports BINARY .gcode, which is not supported by these printers. By default, PrusaSlicer should assume TEXT .gcode format for Creality printers.

Creality Ender-3 S1 Pro is not configured to use binary g.code and it outputs standard g-code.

> Once .gcode output format was manually configured as regular TEXT for my printer, I sliced a 557-layer object and saved the .gcode to SDCard. When I printed the resulting file, the printer seemed to [...]

I don't think its related to PrusaSlicer output. I think your filesystem od USB may be corrupted, which also may let you think that the output is binary.

Anyway, please provide your 3MF project, so we can check if it's configured correctly on your side.

rjflory, 2024-02-13:
New-user/interface issue.  The wizard gave the impression that adding the new printer would also select it- it did not.  So the gcode was output for a different printer than desired.

closed 2024-02-13 by rjflory