prusa3d/PrusaSlicer #12235: Prusaslicer Crashing when USB Plugged In

aaron9412, 2024-02-12:
### Description of the bug

Prusaslicer crashes when USB drive is plugged in:
[image]

This looks to be an issue with prusaslicer trying to write a settings file to the USB. This occurs on a computer that has security settings which I cannot see. Oddly, it happens when any USB flash drive is plugged in, so does Prusaslicer try to write a settings file to every usb? I would be surprised if there was a security block from searching for a setting. Also, as another test related to this issue but unrelated to Prusaslicer, we tried to copy anything with an executable or settings file on to any USB drive and it looks like we cannot.

This issue is fatal and will occur on startup of prusaslicer if a usb drive is plugged in or will occur after having prusaslicer open whenever a usb drive is plugged in.

Thanks!
-Aaron

### Project file & How to reproduce

Run Prusaslicer while a USB flash drive is mounted

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.1

### Operating system

Windows 10

### Printer model

MK4

udaredme, 2024-03-11:
Same for me

hjalte33, 2024-06-20:
I have the same issue on my work computer which has group policies applied. I do not know exactly what causes it. 
Prusa slicer crashes as soon as i plug in a usb flash stick and if the usb is already plugged it crashes as soon as prusaslicer opens up. 
Oddly enough i can use the SD card ready in my pc without issues. This bug only affects usb flash drives. 

if the error is ignored and i leave the dialog box open I can still navigate prusaslicer, but is is really laggy and completely unusable. clicking ok on the error prusaslicer just closes. 

version 2.7.4
Windows 10 
mk3, mk4, prusa XL, prusa mini. Happens to any printer.

hjalte33, 2024-06-27:
update: 
issue still pressent in prusa slicer version 2.8.0

closed 2026-09-29 by prusaslicer-bot[bot]