prusa3d/PrusaSlicer #12240: [FEATURE REQUEST] per object thumbnails

beelsebob, 2024-02-13:
When you print a bunch of instances of an object, it can be hard to tell which object is which and which to cancel. This could be vastly improved if the slicer included thumbnails for each object, that could then be displayed at the printer's cancel object screen.  To do this, PS would need to include thumbnails in the gcode for each object.

The thumbnails should be just like the main thumbnail, except that only on object should be orange. The rest should be greyed out.

beelsebob, 2024-02-13:
Related Buddy Firmware issue - https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3744

Tupson444, 2024-10-25:
This would address one of the biggest issues of Cancel Objects feature: identifying what to cancel without sitting next to the printer waiting for it to start printing the failed object and then use "cancel current".

This would be very useful so I hope it will be worked on soon.

Tupson444, 2026-09-18:
still wanted (some solution for identifying objects)