prusa3d/PrusaSlicer #12242: Variables for Text

hebbet, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
It would be nice to have something like variables in Text, that eg. includes the used material or other print settings

**Describe the solution you'd like**
I suggest shortcuts like {material} or {layerheight} are replaced with the actual settings for that object.
[image]

**Describe how it would work**
with that you can easily stamp in some info when you experiment with different settings, like material, infill etc. 
when you change infill from 20% to 30% you don't have edit the text

**Describe alternatives you've considered**
do it by hand

mix579, 2024-02-13:
Not a bad idea at all.

hebbet, 2026-09-02:
that is a feature request and not a bug report