prusa3d/PrusaSlicer #12246: Unable to Launch GUI  on Wayland

fg-charles, 2024-02-14:
### Description of the bug

When use  `prusa-slicer`, installed from the official arch repo, it refuses to open the GUI and displayes the message: `DISPLAY not set, GUI mode not available.` This is despite having set `GDK_BACKEND=wayland`.
I suspect this is related to the following code:
```
// src/PrusaSlicer.cc:698
#ifdef SLIC3R_GUI
    #if !defined(_WIN32) && !defined(__APPLE__)
        // likely some linux / unix system
        const char *display = boost::nowide::getenv("DISPLAY");
        // const char *wayland_display = boost::nowide::getenv("WAYLAND_DISPLAY");
        //if (! ((display && *display) || (wayland_display && *wayland_display))) {
        if (! (display && *display)) {
            // DISPLAY not set.
            boost::nowide::cerr << "DISPLAY not set, GUI mode not available." << std::endl << std::endl;
            this->print_help(false);
            // Indicate an error.
            return 1;
        }
```
I've tried un-commenting the lines labelled by `<---`, replacing the `if` statement, but this does not work, and results in the following ouput:
```
/home/cgf/.local/src/PrusaSlicer/deps/build/dep_wxWidgets-prefix/src/dep_wxWidgets/src/gtk/msgdlg.cpp(257): assert ""m_widget"" failed in ShowModal(): failed to create GtkMessageDialog

(prusa-slicer:113380): GLib-GObject-CRITICAL **: 16:09:48.070: invalid (NULL) pointer instance

(prusa-slicer:113380): GLib-GObject-CRITICAL **: 16:09:48.070: g_signal_handlers_disconnect_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
```
I'd love to know how this can be fixed / worked around so I can use this software on my daily driver. Thanks!

NOTE: Issue $8284 seemed somewhat related, but it is quite old and riddled with users having segfaults due to faulty arch packaging at the time. Other issues under the search term "wayland" indicate it is possible, at least older versions, but I have not been able to get this to work and am out of ideas. The code above and the ignoring of GDK_BACKEND is how I concluded that it is likely a bug.

### Project file & How to reproduce

Simply run `prusa-slicer` in an Arch x86 machine, with the binary from the `extra` Arch package repositiory.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

version_2.7.1

### Operating system

Arch Linux x86_64

### Printer model

n/a

lukasmatena, 2024-02-15:
>installed from the official arch repo

anarsoul, 2024-02-15:
Likely, you don't have Xwayland configured in your compositor. Prusa Slicer forces X11 backend due to bugs in WxWidgets. The bugs were fixed, but it'll take some time for devs to update to newer WxWidgets.

fg-charles, 2024-02-16:
@lukasmatena No, I have not tried the AppImage. 

@anarsoul I see. I didn't know that I needed XWayland. I'll stay on the lookout for the WxWidget update. Thanks!

closed 2024-02-16 by fg-charles