prusa3d/PrusaSlicer #12247: Multicolor Paint only showing 16 colors

JWSmythe, 2024-02-14:
### Description of the bug

When painting on colors for multi-color printing, only the first 16 colors are presented as options, despite having 20 colors defined.  I tried this with 19 and 20 colors defined, both of which cut off at 16.

I assume this is just a max limit.  It should be 32 or 64, to allow for a decent size pallet that I can always use rather than customizing the virtual tools list for each print job.

[image]

### Project file & How to reproduce

1) Configure PrusaSlicer for > 16 colors. 
2) In Editor view, select the object. 
3) Click the drop list for either First or Second color.  You will only see the first 16 colors there. 

[CYMK_3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/14274241/CYMK_3DBenchy.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+win64

### Operating system

Windows 11 Version 10.0.22622 Build 22622

### Printer model

Ender 5 S1 + BTT Octopus + Zonestar M4V6

Jan-Soustruznik, 2024-02-14:
Hello, @JWSmythe , 

thank you for your report. Currently status(max 16colors) is limited by current architecture.

NielschB, 2024-11-08:
Are there any plans to update the limit to 32 or 64?