prusa3d/PrusaSlicer #12248: Variable layer height uses layer height limits from off-by-one extruder

matthewfallshaw, 2024-02-14:
### Description of the bug

The variable layer height tool appears to be using the Printer Settings > Layer height limits from the [n-1]th extruder (instead of the nth extruder).

### Project file & How to reproduce

My project file (though note a reproduction from stock below):
[variable layer height issue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14277611/variable.layer.height.issue.3mf.zip)

Prusa XL 5 tool
Printer Settings > Extruder 3 > Layer height limits > Max: 0.24mm
[image]

Part set to use Extruder 3, but Variable layer height tool > Adaptive sets a section of the print to layer height 0.3mm (which is larger than the configured maximum layer height for the configured extruder).
[image]

So, that's confusing, but I notice that:
Printer Settings > Extruder 2 > Layer height limits > Max: 0.3mm
[image]

Let's try changing that (extruder 2, the [n-1]th extruder) to 0.24mm:
[image]

Back to the Variable layer height tool and click Adaptive again, and:
[image]

Project file attached, though a repro that doesn't require that project file (using only stock Prusa profiles) follows:
1. New project with:
Printer: Original Prusa XL - 5T Input Shaper 0.6 nozzle
Filament: Probably doesn't matter, but I set them all to Generic PETG
Print Setting: 0.32mm STRUCTURAL
2. Add Shape > Sphere
3. Add Layers and Perimeters to that shape, and remove all but Layer height. Set Layer height to 0.24mm (seems necessary to set a layer height lower than the limit you're going to set, since the Variable layer height tool doesn't seem willing to reduce layer height below whatever the default set for the part).
4. Right-click to set the Extruder for the Sphere to 2.
5. Printer Settings > Extruder 2 > Layer height limits > Max: 0.3mm (note that the default, still retained at this point by Extruder 1, is 0.4mm)
6. Plater > Variable layer height > click the Sphere > Adaptive
7. Note that the middle section of the Sphere is a bright red, illegal 0.4mm, ignoring the Max we set for the selected extruder.
8. Printer Settings > Extruder 1 > Layer height limits > Max: 0.2mm (picking a different value, so we now have 1 on 0.2mm, 2 on 0.3mm, and the rest on 0.4mm)
9. Plater > Variable layer height > click the Sphere > Adaptive
10. Note that the middle section of the Sphere is a nice, boring, ~legal~… okay, so it's now 0.24mm instead of 0.2mm because Variable layer height takes the default layer height we set to 0.24mm above and uses that instead of the Max 0.2mm we set, which isn't ideal, but I'm just going to note that in passing and move along. 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.1+MacOS-arm64

### Operating system

MacOS 14.3.1

### Printer model

Original Prusa XL - 5T

lukasmatena, 2024-02-14:
I am sorry, but I don't understand. There is just too much information. The 3MF you posted uses extruder 3 and 5, the limits are 0.075-0.24 and 0.15-0.4. Adaptive layer height is set between 0.15 and 0.24, none of the limits are violated.

The only way it could be violated is if layer height in Print Settings was set outside the allowed range. In that case the range is expanded. If you think there is some issue, please attach a 3MF which shows the violation of the limits.

lukasmatena, 2024-02-15:
@matthewfallshaw Thanks for the condensed 3MF demonstrating the problem clearly. You are absolutely correct. There is an off-by-one bug when the limits of the extruders are collected. The bug is there for more than 6 years, and it apparently went unnoticed because most people use the same limits for all the extruders. We already have a fix, which will be included in the upcoming 2.7.2-beta1.

lukasmatena, 2024-02-20:
Fixed in [2.7.2-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-beta1). Closing.