prusa3d/PrusaSlicer #12249: Option for "Hilbert Curve" as solid infill patern

Lichtjaeger, 2024-02-14:
**Is your feature request related to a problem? Please describe.**
Hi, I'm currently printing with PA. I have a solution for the bed adhesion problem. But the part I need to print has overhangs and bridges. Those are the places where my print is failing due to warping.

I'm already using common anti-warping settings like reduced layer height, draft shield, slow print temperatures and no fan. But I think there is still something that would help.

**Describe the solution you'd like**
In this video of CNC-Kitchen (timestamp 7:38), I saw that the solid infill pattern does influence the warping of parts.

https://youtu.be/TGa_KvKLDR8?t=458&si=rdMkvX0R0MkgfT3g

So, an option to change the solid infill from Rectilinear to "Hilbert Curve", as you can already do for the bottom and top layer, might be beneficial for printing with materials that tend to warp like Nylon.

**Describe how it would work**
[Here is the Link](https://www.researchgate.net/publication/313685481_Fractal_Scan_Strategies_for_Selective_Laser_Melting_of_'Unweldable'_Nickel_Superalloys) to the research document.

**Describe alternatives you've considered**
Setting the top and bottom layers to 1, the infill pattern to Hilbert Curve and the infill density to 100% would allow for a similar behaviour. But that would waste a lot of filament and time.

MC42, 2024-03-13:
This is obviously imperfect as your original feature request, but it appears you can set the infill to be 99.9999% and avoid the 100% -> automatic swap to rectilinear.  

This may help with proving the concept; best of luck :)

rcloran, 2025-01-06:
This looks like a duplicate of #2147

closed 2026-09-29 by prusaslicer-bot[bot]